arduino/Arduino #6087: How does a library module determine the hardware

rrvt, 2017-03-17:
How does a library determine the hardware being used.  By this I mean the processor and the Arduino/Other board.  If one chooses to develop a library that writes things close to the hardware it is essential that the correspondence between chip registers and the external pins on the board be known at compile time so that different choices may be made at compile time.

An ino file is compiled with one of the following identifiers defined:
 __AVR_ATmega328p__
__AVR_ATmega168__ 
__AVR_ATmega1280__
__AVR_ATmega2560__
__AVR_ATmega32u4__
__AVR_ATtiny85__  

Would it be possible to do the same for the library files?

It probably would be useful to also define an identifier for the specific board also so that mappings between the internal registers and external pins could be defined...

facchinm, 2017-03-20:
Sure, you can write processor-specific code by enclosing the relevant bits between `ifdef`s as described here: http://www.atmel.com/webdoc/avrlibcreferencemanual/using_tools_1using_avr_gcc_mach_opt.html

For example, a piece of code specific for the UNO will look like
```
#ifdef __AVR_ATmega328P__
...code here
#endif
```

closed 2017-03-20 by facchinm

cmaglie, 2017-03-20:
There are also a specific define per-board, for example the Arduino Leonardo has:

```
ARDUINO_AVR_LEONARDO
ARDUINO_ARCH_AVR
```

you can quickly find them if you enable "verbose compile" on the preferences:

```
     [exec] "/home/cmaglie/Code/arduino/build/linux/work/hardware/tools/avr/bin/avr-g++"
-c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections
-fno-threadsafe-statics -MMD -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10802
-DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR     <-----
-DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"'
'-DUSB_PRODUCT="Arduino Leonardo"'
"-I/home/cmaglie/Code/arduino/build/linux/work/hardware/arduino/avr/cores/arduino"
"-I/home/cmaglie/Code/arduino/build/linux/work/hardware/arduino/avr/variants/leonardo"
"/home/cmaglie/Code/arduino/build/linux/work/hardware/arduino/avr/cores/arduino/main.cpp"
-o "/tmp/arduino_build_72059/core/main.cpp.o"
```

rrvt, 2017-03-20:
Both solutions work, thanks.  Bob