arduino/Arduino #6088: Sketch doesn't always refresh when the Arduino IDE from an external editor

Xorgon, 2017-03-18:
When using the Arduino IDE on Windows with an external editor (set as such in the preferences) simply clicking back onto the window doesn't refresh the sketch that is open. As far as I am aware, it must be refocused by clicking on its icon in the toolbar to get it to refresh the sketch.

facchinm, 2017-03-20:
Hi @Xorgon ,
would you mind testing https://github.com/arduino/Arduino/pull/6074 ? It should reload the sketch also when you are not in "external editor mode" when you refocus the IDE window.

Xorgon, 2017-03-20:
@facchinm I've had a go using that build. It has improved the problem, but there's still at least one situation where it doesn't reload correctly, although from my observation of its behavior I think it may now be limited by the editor saving it at certain points. I've recorded what happens, (time-stamped links are also below):
[https://youtu.be/GWvs3CixWI4](https://youtu.be/GWvs3CixWI4)

The behaviors that the video demonstrates are as follows:
- At [0:06](https://youtu.be/GWvs3CixWI4?t=6s): Adding to a file does not cause Arduino to reload the file, unless the Arduino window is clicked into at least twice (an improvement from having to minimize and maximize the window).
- At [0:22](https://youtu.be/GWvs3CixWI4?t=22s): Removing from a file does cause Arduino to reload the file (with an error).
- At [0:28](https://youtu.be/GWvs3CixWI4?t=28s): Adding to a file will cause Arduino to reload the file if the toolbar is clicked (hence my thoughts on it now being a limitation on the editor side). This also throws errors.

The stacktrace text is [here](https://gist.github.com/988fa31aabf2155969bccb1ca2f467f6).

facchinm, 2017-03-20:
PR #6074 doesn't handle the full file wipe ATM, and I think it shouldn't (or it should show an informative popup). The double click behaviour is quite strange (it doesn't happen on Linux so it deserves more investigation).
After this roundup, it looks like the problem is related with how Windows10 implements the "reacquire focus" action, since the code is shared with platforms that work consistently.

facchinm, 2017-03-20:
Mmmmh, I just tested on a Win10 VM and I can't reproduce your behaviour with plain 1.8.1... Could it be a Java bug related with the particular video card (like Nvidia's bug some times ago)?

Xorgon, 2017-03-20:
Hmmm, odd, the only other factor I can think of would be the save behavior of CLion (the Jetbrains IDE for C/C++).

ckunta, 2018-04-22:
I found out that the behavior of not refreshing still being exhibited in Arduino 1.8.5. I'm in Windows 10 and use Emacs as external editor. Although, "Fix Encoding and Reload" in Tools seems a reasonable workaround to manually reload it.