arduino/Arduino #6093: struct lconv not declared in locale.h

KarlZeilhofer, 2017-03-19:
In a library I'm developing, I'm using the function `localeconv()` from `locale.h`

in Arduinos `locale.h` the `struct lconv` is only forward declared, but nowhere defined. 

The compiler complains about `incomplete type 'struct lconv' `
```
/home/karl/Arduino/libraries/Snaiks/snproperty.h:106:32: error: invalid use of incomplete type 'struct lconv'
       TempStr[i] = localeconv()->decimal_point[0];
                                ^
.../arduino-1.6.12/hardware/tools/avr/avr/include/locale.h:12:8: error: forward declaration of 'struct lconv'
 struct lconv;
        ^
```

matthijskooijman, 2017-03-19:
This is an issue with avr-libc, not Arduino itself. AFAIK avr-libc does not contain any locale information or support, so I'm already surprised that the `localeconv()` is declared at all. If you want to do locale-dependent stuff, I believe you'll have to resort to some Arduino-specific library for that (if one exists, I haven't checked).

oqibidipo, 2017-03-19:
The locale support is just a stub:

http://svn.savannah.gnu.org/viewvc/avr-libc/trunk/avr-libc/include/locale.h?view=log
http://svn.savannah.gnu.org/viewvc/avr-libc?view=revision&revision=2427
> This provides a bit of functionality, but mostly stubs, for hosted features
that we've been missing.  The alternative would be to make avr an outright
unhosted target, but that'd make it harder to test, and would also stop
using some applications that depend on minimal hosted features.

facchinm, 2017-03-20:
As pointed out by @matthijskooijman and @oqibidipo , avr-libc comes with no locale support, as most C libraries for OS-less targets (if we exclude semihosting scenarios or similar), so there is nothing we can do unfortunately :disappointed:

closed 2017-03-20 by facchinm