arduino/Arduino #6095: toggle();

Testato, 2017-03-19:
many core have a toogle() function.
Also some third part arduino core have it.
a toogle() function is useful on many or all project, 
please add it to the official arduino core.

PaulStoffregen, 2017-03-20:
> Also some third part arduino core have it.

Any chance you could be more specific about which ones?

MauroMombelli, 2017-03-20:
It is a XOR on the register with a mask forma the pin, differenti core are forma different architecture and may have different and specific application. Notice this will also work for enable/disable pull up.

This is the classic simple and quick modification for the core, and uneccessary hard for the newbie.

Testato, 2017-03-20:
I do not remember in wich core i view it, but is there also this lib
http://playground.arduino.cc/Code/DigitalToggle

Testato, 2017-03-20:
@PaulStoffregen 
I do not remember in wich arducore i view it, but is there also this lib
http://playground.arduino.cc/Code/DigitalToggle

the code in it is very simple:
```
void digitalToggle(uint8_t P)
{
  *portInputRegister(digitalPinToPort(P)) = digitalPinToBitMask(P);
}
```

cmaglie, 2017-03-20:
Please do not post useless comments with only "+1" in it, there is a specific function of github for doing so that adds "likes" to the comment **without** spamming the mailbox of 800+ people.

Thanks.

cmaglie, 2017-03-20:
#111

cousteaulecommandant, 2017-04-12:
By the way, no need to use XOR.  ATmega already provides toggle functionality by writing a 1 to the PINx registers (the ones normally used for reading), so I guess it'd be a good idea to use that.

5N44P, 2017-04-12:
@cousteaulecommandant exactly. 
In fact the code posted by @Testato from the digitalToggle library uses this feature. 

We want this feature 😄😄

Testato, 2017-04-12:
In this lib is there also a very useful ToggleMillis()
It toggle a pin using millis() so whitout blocking delay.
https://github.com/Testato/ToggleMillis

cousteaulecommandant, 2017-04-18:
@5N44P Ah, right, I missed that -_-"
(In my defense, I was talking about the code from PR #111 which cmaglie referenced; and also the mention of XOR by MauroMombelli; I didn't notice Testato's snippet)

5N44P, 2017-08-30:
Cool, it has been labeled! This means that it will be considered? It would be nice if someone from arudino could explain if and how this type of requests are handled. 
Thank you!

Testato, 2019-03-23:
ST added Toggle function to the ST Official Arduino Core

```
digitalToggle	KEYWORD2
digitalToggleFast	KEYWORD2

```