arduino/Arduino #6102: SPI.h Libuary

DevilWAH, 2017-03-20:
OK I have a issue with this Library

I have code something like this to control a APA102 LED addressable strip. 

```
void lightLED(Value)
{

spi.transfer(&value, 4); //I know that value will be a uint32_t value and changed this in testing from sizeoff()
Serial.print(value);

}

uint32_t LED = 0x00000000;
lightLED(LED);

LED = 0xE1FF0000;
lightLED(LED);

//LED = 0xE10000FF;
//lightLED(LED);
```

So this avoided the buffer overwrite issue i was having as i am re-writing the variable each time. 

and the format for the LED strip is 

1. one start from of 32 zeros
2. n number of control frames of 4 X 8bits. the first of these request the first 3 bits set to 1 and the reaming 5 provide 31 levels of brightness. and then Blue/Green/Red take up the next 3 lots of 8 bits

So if I run the code above with the lines commented out I get a blue LED, however if i uncomment the second LED I end up with single Purple LED. A serial write in the function output the correct decimal value, but it does not seem the SPI output is as expected. 

Could the Library allow single bit transfers and arbitrary length transfers , as looking at the library even transferring a buffer still  does so in 8bit chunks.

facchinm, 2017-03-21:
Hi @DevilWAH ,
4 single bytes consecutive transfers are indistinguishable from a single 4 bytes transfer (as long as they are atomic, thus enclosed in `beginTransfer` and `endTransfer`).
The behaviour you are experiencing probably comes from the particular LED strip you are using; I'd suggest you to ask the [forum](http://forum.arduino.cc/) for assistance in this case.
Thanks

closed 2017-03-21 by facchinm

PaulStoffregen, 2017-03-21:
This code is incorrect, in many ways.  It does not even compile!  Ask for help on the forum to fix it.

    void lightLED(Value)
    {
    
    spi.transfer(&value, 4); //I know that value will be a uint32_t value and changed this in testing from sizeoff()
    Serial.print(value);
    
    }

PaulStoffregen, 2017-03-21:
Ask for help on the forum.  When you post on the forum, please do not waste people's time with wrong code.