arduino/Arduino #6109: TabControl can't deal with much tabs

julianwap, 2017-03-22:
Hi,

I'm using the Arduino IDE for a very large project (31 seperate Sketch files).
Switching through these tabs with the context menu on the right is working but it's not possible to scroll right inside the tab control as it is possible in other tab controls.

Because the scrolling is not implemented and I had to work with 31 tabs, I used my three monitors and resized the window to a width of 3600 pixels and noticed that the header has a maximum width of 3000 pixels and is then leaving ugly padding spaces on the left and right sides. (see screenshot)

[image]

Thanks for your fantastic work on Arduino,
Julian

per1234, 2017-07-18:
>the header has a maximum width of 3000 pixels and is then leaving ugly padding spaces on the left and right sides.

This part was fixed in https://github.com/arduino/Arduino/commit/ddf60361d6a1ffec6c5e6c064ad0e9f174104704
>it's not possible to scroll right inside the tab control as it is possible in other tab controls.

This part is very similar to https://github.com/arduino/Arduino/issues/1193 but maybe not quite a duplicate?

Also related (but not duplicate): https://github.com/arduino/Arduino/issues/3601

julianwap, 2017-07-18:
> This part was fixed in ddf6036
Perfect

> This part is very similar to #1193 but maybe not quite a duplicate?
There are different approaches for this problem, I'd be happy with scrolling or multiline tabs.