arduino/Arduino #6116: Cannot install arduino-1.8.2

fturco, 2017-03-24:
I'm trying to install arduino-1.8.2-linux64 under a Gentoo Linux system but when running the `install.sh` script I get the following error messages:

```
$ ./install.sh 
Adding desktop shortcut, menu item and file associations for Arduino IDE...touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/fturco/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
xdg-mime: file '/home/fturco/Buffer/arduino-1.8.2/lib/cc.arduino.arduinoide.xml' does not exist
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-desktop-menu: filename 'cc.arduino.arduinoide.desktop' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example filename is 'example-cc.arduino.arduinoide.desktop'
Use --novendor to override or 'xdg-desktop-menu --manual' for additional info.
xdg-desktop-icon: filename 'cc.arduino.arduinoide.desktop' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example filename is 'example-cc.arduino.arduinoide.desktop'
Use --novendor to override or 'xdg-desktop-icon --manual' for additional info.
xdg-mime: file '/home/fturco/Buffer/arduino-1.8.2/lib/cc.arduino.arduinoide.xml' does not exist
 done!
```

There's no Arduino IDE among my applications after the previous installation process.

Please notice I can install the previous Arduino version (1.8.1) without any problems.

mjhowell, 2017-03-25:
Pretty similar result on Ubuntu 16.04:

```
$ ./install.sh 
Adding desktop shortcut, menu item and file associations for Arduino IDE...xdg-mime: file '/home/markh/SDKs/arduino-1.8.2/lib/cc.arduino.arduinoide.xml' does not exist
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-desktop-menu: filename 'cc.arduino.arduinoide.desktop' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example filename is 'example-cc.arduino.arduinoide.desktop'
Use --novendor to override or 'xdg-desktop-menu --manual' for additional info.
xdg-desktop-icon: filename 'cc.arduino.arduinoide.desktop' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example filename is 'example-cc.arduino.arduinoide.desktop'
Use --novendor to override or 'xdg-desktop-icon --manual' for additional info.
xdg-mime: file '/home/markh/SDKs/arduino-1.8.2/lib/cc.arduino.arduinoide.xml' does not exist
 done!
```
1.8.1 and earlier with install.sh script worked fine.

per1234, 2017-03-25:
Related pull request: https://github.com/arduino/Arduino/pull/6110

hmattiasl, 2017-03-29:
I had the same problem while installing Arduini IDE 1.8.2 on a Ubuntu machine. The resource name in the install.sh file is wrong. When I looked in the Arduino-1.8.2/lib there is a file named arduino-arduinoide.xml file. So I changed the "RESOURCE_NAME=cc.arduino.arduinoide" to "RESOURCE_NAME=arduino-arduinoide" and then run ./install.sh and that worked fine for me.

jerryneedell, 2017-03-30:
I had the same issue with the linuxarm distribution on a Raspberry Pi.
The same fix worked for me as well.

theshadow124, 2017-04-01:
hmattiasl's solution fixed it for me too.

johnmoreland, 2017-04-03:
installing arduino IDE 1.8.2 on ubuntu 14.04. hmattiasl's solution worked on my machine.

supertin, 2017-04-04:
Just adding that it's the same problem and fix in the 32 bit version...

2ra3, 2017-04-05:
thanks hmattiasl.

gilfagote, 2017-04-07:
Hi,
I've tried to use the same solution and I still have the same problem.

Adding desktop shortcut, menu item and file associations for Arduino IDE...xdg-mime: file '/opt/arduino-1.8.2/lib/arduino.arduinoide.xml' does not exist
xdg-icon-resource: icon name 'arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-desktop-menu: filename 'arduino.arduinoide.desktop' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example filename is 'example-arduino.arduinoide.desktop'
Use --novendor to override or 'xdg-desktop-menu --manual' for additional info.
xdg-desktop-icon: filename 'arduino.arduinoide.desktop' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example filename is 'example-arduino.arduinoide.desktop'
Use --novendor to override or 'xdg-desktop-icon --manual' for additional info.
xdg-mime: file '/opt/arduino-1.8.2/lib/arduino.arduinoide.xml' does not exist
 done!

jerryneedell, 2017-04-07:
It looks like you just removed the cc.
but did not change arduino to arduino-arduinoide
From the posted fix:

`So I changed the "RESOURCE_NAME=cc.arduino.arduinoide" to "RESOURCE_NAME=arduino-arduinoide"`

I hope that helps!

gilfagote, 2017-04-07:
Thanks, jerryneedell!
I'm newbie in Arduino and programming...

Again, Thanks!!

anial-pro, 2017-04-08:
Thanks, jerryneedell!
This work.
XUbuntu 16.04

KahYee940, 2017-04-09:
thanks for the solution @hmattiasl

achimboers, 2017-04-09:
For those of you just a dyslectic as me....
Note the dash instead of the dot between arduino and arduinoide.
Not only remove just the cc. 
Also replace the . for a - 
Than it works on the Raspberry Pi

cmaglie, 2017-04-11:
Fixed by #6110 the patch will be availble on the next release 1.8.3

closed 2017-04-11 by cmaglie

amark14912, 2017-04-22:
hmattiasl's solution fixed it for me too. Using Ubuntu 16.04

elalexluna83, 2017-04-27:
Thanks hmattiasl.. Your workaround worked like a charm for me.

harishkumarr95, 2017-05-02:
In my .xml file there was no RESOURCE_NAME given. The contents of my file are just,

eracoon, 2017-05-06:
@hmattiasl for president, thank you for the solution... it worked perfectly

elpluto, 2017-05-07:
@hmattiasl for developer of the year. Work for me too. Thanks

jediofgever, 2017-05-20:
Editing install.sh file i get över this issue. But now the arduino İde icon raises error when i try to open İDE. The error says: invalid desktop entry file arduino-arduino

What can be the cause of this problem?
İ am running on Raspbian Os on pi 3

jerryneedell, 2017-05-20:
@jediofgever Check the settings for the Icon.
Do a right click and select "properties" then look in the Desktop Entry Tab and verify thath the Command points to wherever you unpacked the 1.8.2 distribution.

jediofgever, 2017-05-20:
@jerryneedell  my extracted  arduino-1.8.2  file is under /opt directory of linux file system. Although I pointed Command tab to that directory  the problem still remains

gsldynamics, 2017-05-28:
@hmattiasl you saved me a big headache!  Worked great!

per1234, 2017-05-28:
@facchinm it seems like the IDE has not been under active development lately. Why not do a release so people can stop running into this bug? I don't see the point in waiting 2 months after a serious bug is identified and fixed.

stmic999, 2017-05-28:
arduino-1.8.2 installation failed on my ubuntu16.04.2 desktop version. It looks pretty much like the configuration of the 1-8.2 is not correctly set before released. Please check it. Thanks.

facchinm, 2017-05-29:
Hi @per1234 ,
I'm sorry for the long hiatus and I'm sad that from the outside it appears that we are not interested in publishing a point release to fix this bug. The reason is, we are crazily busy these days, and the features we wanted to integrate in the upcoming IDE are huge but need some more work to be taken, so spending a couple of days to validate a new version is not feasible at the moment. Could it make sense to release a Linux-only, untested, 1.8.2r1 only containing https://github.com/arduino/Arduino/commit/bd4921910987bf9ec1c19ed2e9fd1c966180282d over 1.8.2 tag?

cmaglie, 2017-05-29:
Using tags like 1.8.2.r1 have made a lot of confusion in the past, at this point let's make a "service" release 1.8.3.

per1234, 2017-05-29:
Yeah, stick with semver. That would only allow for a 1.8.3-rc1 type of versioning as 1.8.2-r1 would be considered a pre-release to 1.8.2, which is definitely wrong. I don't see any problem with just bumping the patch version, even for a release with only minimal changes.

terxw, 2017-06-02:
Works for me too!

rohit403, 2018-02-15:
# installed arduino in arch linux using .tar file. but while uploading sketch it is showing this error. how to to fix it?

[alarm@alarmpi arduino-1.8.5]$ ./install.sh 
Adding desktop shortcut, menu item and file associations for Arduino IDE...touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
touch: cannot touch '/home/alarm/.local/share/icons/hicolor/.xdg-icon-resource-dummy': No such file or directory
 done!

Sketch uses 444 bytes (1%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.

An error occurred while uploading the sketch

/home/alarm/Downloads/arduino-1.8.5/hardware/tools/avr/bin/avrdude: error while loading shared libraries: libtinfo.so.5: cannot open shared object file: No such file or directory

facchinm, 2018-02-15:
@rohit403 the problem you are encountering is due to a missing dependency on the ARM build. Simply install `aur/libtinfo5` and the error should be gone

rohit403, 2018-02-16:
Thank you sir... it works

dwisatriow, 2018-10-21:
@hmattiasl Thank you. In my case i can't find resource name in arduino-arduinoide.xml, but then i try to change appdata.xml `<id>cc.arduino.arduinoide.desktop</id>` to `<id>arduino-arduinoide.desktop</id>` then run ./install.sh and the error seems gone. Is it ok if i'm doing that?

lorenzolightsgdwarf, 2018-11-11:
It's in  appdata.xml

ravi312, 2019-10-16:
thanks brenly. The resource name is correct in the arduino.1.8.10 but the error is in the appdata.xml file which resides in lib folder. Changing cc.arduino.arduinoide.desktop to arduino-arduinoide.desktop, works for me too!

Lappen2108, 2020-06-24:
Hey. I want to install Arduino IDE on Raspberry Pi 3 b +. I tried all of your solutions, but nothing hapend. When I install, i get this error: 

Adding desktop shortcut, menu item and file associations for Arduino IDE...

touch: cannot touch '/root/.config/mimeapps.list': No such file or directory
/usr/bin/xdg-mime: 848: /usr/bin/xdg-mime: cannot create /root/.config/mimeapps.list.new: Directory nonexistent

 done!

and if I lunch the IDE, i get this error: 

nvalid desktop entry file: '/usr/share/applications/arduino-arduinoide.desktop'

thank you

Lappen2108, 2020-06-24:
Hooray, it works! I did it with the script from [scruss](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=194199&p=1283801#p1283757 ) Now I only have Arduino 1.8.5, but better than nothing

ErrorBot1122, 2021-01-24:
is this still relevant?

Lappen2108, 2021-01-24:
no! I do‘nt try this anymore.. You could close this🙂