arduino/Arduino #6118: cannot install arduino-1.8.2 ide

thesauband, 2017-03-25:
I got the following err msg
bandula@bandula-Latitude-13:~/Downloads/arduino-1.8.2$ ./install.shAdding desktop shortcut, menu item and file associations for Arduino IDE...xdg-mime: file '/home/bandula/Downloads/arduino-1.8.2/lib/cc.arduino.arduinoide.xml' does not exist
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-icon-resource: icon name 'cc.arduino.arduinoide' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example icon name is 'example-cc.arduino.arduinoide'
Use --novendor to override or 'xdg-icon-resource --manual' for additional info.
xdg-desktop-menu: filename 'cc.arduino.arduinoide.desktop' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example filename is 'example-cc.arduino.arduinoide.desktop'
Use --novendor to override or 'xdg-desktop-menu --manual' for additional info.
xdg-desktop-icon: filename 'cc.arduino.arduinoide.desktop' does not have a proper vendor prefix
A vendor prefix consists of alpha characters ([a-zA-Z]) and is terminated
with a dash ("-"). An example filename is 'example-cc.arduino.arduinoide.desktop'
Use --novendor to override or 'xdg-desktop-icon --manual' for additional info.
xdg-mime: file '/home/bandula/Downloads/arduino-1.8.2/lib/cc.arduino.arduinoide.xml' does not exist
 done!

per1234, 2017-03-25:
Duplicate of https://github.com/arduino/Arduino/issues/6116

facchinm, 2017-03-27:
Closing as duplicate of  #6116

closed 2017-03-27 by facchinm