arduino/Arduino #6128: editor.indent setting in 'preferences.txt' has NO effect in 1.8.1

bombasticbob, 2017-03-27:
Because it is SO irritating (like it also was for THIS guy:  http://forum.arduino.cc/index.php?topic=247973.0 ), I want to SHUT OFF any "help" the IDE *FEELS* I need when typing things.

This PARTICULARLY includes ANY form of "autocomplete".

I type SO fast, I don't NEED it.  In fact, my 'finger memory' assumes it WILL NOT BE THERE.

So, when it _IS_ there, it's an EXTREME IRRITANT.

Well, I tried the solutions offered in the above forum link, that is to set 'editor.indent' to FALSE.

I did this in the 'preferences.txt' in '.arduino15' and ALSO the one in the 'lib' directory [where I'd modified the colors to be SANE so I could actually READ the errors and warnings in the build output] so I'm pretty sure I was adjusting the CORRECT FILES.

And the result?  NOTHING!  I _STILL_ get that EXTREMELY IRRITATING BEHAVIOR.

So, PLEASE FIX the ability to SHUT IT OFF.  Apparently, it USED to work (shutting it off I mean).  And please don't say "use a different editor".  That's not the point.  The point is to FIX THIS ONE.

facchinm, 2017-03-27:
Fixed with #6129. Next time, please avoid using caps lock and this harsh tone or the issue will be closed automatically, no matter how much right you are. If you dislike the IDE so much, there are thousands of alternatives out there, just pick one and enjoy it (or, if you like our IDE in the end, try being more polite please :wink: )

bombasticbob, 2017-03-27:
thanks for getting a test-build out there quickly.

bombasticbob, 2017-03-27:
the test build in #6129 appears to fix the general problem; that is, the auto-indent behavior is now controlled by the editor.indent true/false assignment.  _However_, this does not address my reason for having created this issue in the _first_ place.  As such, I shall create a _new_ issue describing the nature of the problem in full, and recommend the addition of a new setting to control it.  Editing 'preferences.txt' is acceptable to me, which _should_ make this a minimal change.  I may even do it myself with a pull request, if I have time.

bombasticbob, 2017-03-28:
related to this issue, I found the code that inserts the ending curly brace, see latest comment added to #6132 (with recommended change)

closed 2020-12-24 by per1234