arduino/Arduino #6131: GSM3SoftSerial supports only 20, 16 and 8MHz processors

stnorthern, 2017-03-27:
I get an error when I compile with GSM library with my board:

 #error This version of GSM3SoftSerial supports only 20, 16 and 8MHz processors

I have a 48 MHz board

facchinm, 2017-03-27:
Which board are you using? GSM library is only compatible with AVR boards and none of them is clocked at 48MHz. If you are using an Arduino Zero or similar, please reopen this issue in the correct repo (https://github.com/mattairtech/ArduinoCore-samd). Thanks

closed 2017-03-27 by facchinm

stnorthern, 2017-03-27:
I see. 
I am using SmartEverything board. It is Atmel ARM M0+ MCU. 
I am totally new with Arduino, and not even sure if this is AVR?

facchinm, 2017-03-27:
Yes, it's a SAMD boatd (no AVR). The repository is https://github.com/ameltech/sme-arduino-core, please reopen the issue there, thanks!

stnorthern, 2017-03-27:
Okay, I will. 
I have been looking at the requirements for the GSM modem, but I did not find anywhere that AVR was required, nor that only certain frequencies was supported.

stnorthern, 2017-03-27:
The board you linked was for SmartEverything. I dont see why they have anything to do with the GSM modem code (I bought the modem separately). 
I guess I have to make a port for non-avr Arduinos or find another modem? Would it be difficult to make a non-avr port of the code?

facchinm, 2017-03-27:
For this kind of questions you should write to the relevant repo (https://github.com/ameltech/sme-arduino-core) or in the [forum](forum.arduino.cc). This repo tracks problem with the IDE and the AVR core it contains, no other kind of problem.
Probably you will be luckier with [this GSM library](https://github.com/arduino-org/Arduino/tree/master/hardware/arduino/samd/libraries/GSM) but I can't help more than that, sorry.