arduino/Arduino #6132: IDE {} 'autocomplete' behavior in editor needs to be "disable-able"

bombasticbob, 2017-03-27:
In the Arduino 1.0.5 environment, which I've used for a long time, there is no 'autocomplete' behavior for things _like_ adding a '}' 2 lines below where you are when you type '{' followed by a carriage return.

For fast-typists like me, and like <a href="http://forum.arduino.cc/index.php?topic=247973.0">*this*</a> poster, it's *extremely* *irritating* to have a hard-coded "feature" like this, getting in the way _all_ of the time.  And _not_ being able to *turn* *it* *off* is _extremely_ _frustrating_.

A simple solution would be to include a new setting:
<pre>
  editor.autocomplete=false
</pre>
that would disable *all* such featues.  That way, the editor would behave as it did in 1.0.5 without getting in the way, and for those who like these features, it could be "on by default".

ref issues #6128 and #6129

bombasticbob, 2017-03-28:
the problem seems to be in the editor class 'RSyntaxTextArea'

The key to solving this might be here:

https://github.com/bobbylight/RSyntaxTextArea/blob/master/src/main/java/org/fife/ui/rsyntaxtextarea/RSyntaxTextAreaEditorKit.java#L1495

the idea would be to override the necessary class functions and stop this from happening if the 'editor.autocomplete' (or similar) variable is 'false'.

bombasticbob, 2017-03-28:
looks like the correct link for Arduino is this one:

https://github.com/arduino/RSyntaxTextArea/blob/master/src/main/java/org/fife/ui/rsyntaxtextarea/RSyntaxTextAreaEditorKit.java#L1495

the actual code block for 'possiblyCloseCurlyBrace' is here:

https://github.com/arduino/RSyntaxTextArea/blob/master/src/main/java/org/fife/ui/rsyntaxtextarea/RSyntaxTextAreaEditorKit.java#L1561

A test for something _like_ 'editor.autocomplete=false' could be performed in the 'if' statement on line 1575.  This would solve EVERYTHING.

matthijskooijman, 2017-03-28:
@bombasticbob, since the most recent Arduino version, the upstream version of RSyntaxTextArea is used again.

@facchinm, perhaps you could add a note to the https://github.com/arduino/RSyntaxTextArea repository description that it is no longer used as of 1.8.2?

matthijskooijman, 2017-03-28:
Sounds like this should return false: https://github.com/arduino/RSyntaxTextArea/blob/master/src/main/java/org/fife/ui/rsyntaxtextarea/RSyntaxTextAreaEditorKit.java#L1566

Which is done using: http://javadoc.fifesoft.com/rsyntaxtextarea/org/fife/ui/rsyntaxtextarea/RSyntaxTextArea.html#setCloseCurlyBraces%28boolean%29

bombasticbob, 2017-03-28:
the 'setCloseCurlyBraces' function is probably the right way to do it.  thanks.

bombasticbob, 2017-04-05:
verified; see pull request #6155 (tested, works)

per1234, 2017-07-06:
Fixed by https://github.com/arduino/Arduino/pull/6129.

Many thanks to bombasticbob, facchinm, matthijskooijman, and cmaglie for making this happen!!!

closed 2017-07-06 by per1234