arduino/Arduino #6137: readStringUntil - unclear about the return value

jzaruba, 2017-03-30:
https://www.arduino.cc/en/Serial/ReadStringUntil
It is not clear whether the returned value contains the terminator or not. (I read elsewhere that it does not.) And what happens to the terminator? Is it returned in the next Serial.read or is it thrown away?

inflector, 2017-12-05:
From testing the code, the answer to this question is:

1) It does NOT return the terminator.
2) It does NOT leave it in the buffer and return it the next time.

per1234, 2018-09-30:
I'm merging this issue to https://github.com/arduino/reference-en/issues/446 since that is now the more appropriate place for it.

If anyone wants to submit a pull request to https://github.com/arduino/reference-en to clarify the documentation that would be welcome.

closed 2018-09-30 by per1234