arduino/Arduino #6139: Error compiling

ggg-001, 2017-03-30:
Arduino: 1.8.1 (Windows 10), "Arduino/Genuino Uno"

C:\Users\Evgenii\AppData\Local\Temp\ccuTSvlI.s: Assembler messages:

C:\Users\Evgenii\AppData\Local\Temp\ccuTSvlI.s:2308: Error: register number above 15 required

lto-wrapper: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1

cmaglie, 2017-03-31:
can you post the **full** sketch? have you tried with the latest Arduino 1.8.2?

ggg-001, 2017-03-31:
Arduino 1.8.2 no error

akkatracker, 2017-04-01:
So this can be closed?

closed 2017-04-02 by ggg-001

AlexKeisy, 2018-05-24:
Hi Everybody!

This error corresponding on declaration of used registers. 
[image]
in this example "hight"  declared as any register, but used in SUBI operator. 
This example may be compiled right accidentally by one version of compiler(/source) , and not compiled in other case.
Rigth version is:
[image]

AlexKeisy, 2018-05-24:
Hi Everybody!

This error corresponding on declaration of used registers. 
[image]
in this example hight  declared as any register, but used in SUBI operator. 
This example may be compiled right accidentally by one version of compiler(/source) , and not compiled in other case.
Rigth version is:
[image]

Makuna, 2024-04-16:
I recieved this error lately.  It was not helpful to point me to the lines of code that were problematic; but since there was some inline assembly I suspected that was the issue.

The solution was to constrain the variables used more specifically.  Instead of just using "r" for a constraint, use "l", "d", or "a" based on the asm operations used with said variables.  This fixed the problem. 

https://onlinedocs.microchip.com/pr/GUID-317042D4-BCCE-4065-BB05-AC4312DBC2C4-en-US-2/index.html?GUID-E152F8C1-EEE2-4A9D-A728-568E1B02F740