arduino/Arduino #6140: Backing up flash from Arduino Uno and other AVR-based variants [feature request]

DRSDavidSoft, 2017-04-01:
**Introduction**
I think a useful and relatively easy-to-implement feature for the Arduino IDE is the ability to create a backup of the AVR-based boards' flash memory.

For example, if I want to temporarily flash my Arduino with another sketch, this will come extremely handy.
Possible other reasons could be saving the memory of an Arduino, if we've lost the source code, but the HEX will just do for flashing more Arduinos.

**Example**
The IDE already has an option to "Export the sketch", I say we expand this functionality to "Upload HEX file" and "Backup HEX from device memory".

The included `avrdude` in the IDE already supports this, and adding a menu in IDE doesn't seem like a hassle.

Here's a Windows batch file I create to back-up and restore the flash memory of my Arduino:
(The same script can be written for bash or equivalent)

```
@Echo off & cls
cd %0\..

SET AVR_DIR=C:\development\Arduino\arduino-1.x.x\hardware\tools\avr
SET FLASH_FILE=%cd%\flash.hex

REM The following will back-up the flash memory into "flash.hex"
"%AVR_DIR%/bin/avrdude" -C "%AVR_DIR%/etc/avrdude.conf" -v -p atmega328p -c arduino -P \\.\COM8 -b 115200 -D -U flash:r:"%FLASH_FILE%":r 

REM The following will restore (upload) the flash memory from the file "flash.hex"
"%AVR_DIR%/bin/avrdude" -C "%AVR_DIR%/etc/avrdude.conf" -v -p atmega328p -c arduino -P \\.\COM8 -b 115200 -D -U flash:w:"%FLASH_FILE%":r 

pause

```

**TL;DR:** Adding a feature in the IDE to back-up and restore the flash memory residing on the AVR-based Arduino devices.

closed 2021-04-06 by per1234

DRSDavidSoft, 2021-04-06:
@per1234 So, won't add this feature request?

Psychokiller1888, 2021-05-01:
This is just the most usefull thing ever.... And it's crucially missing from the Arduino IDE. Yet another reason I don't use that IDE. Thanks @DRSDavidSoft for the batch file, was too lazy to find out myself how to backup the firmware of my device before messing it up!

ladyada, 2021-05-01:
fyi leaving for future historians - https://github.com/mmoskal/uf2-uno is a neat solution to this

jpl-pfungst, 2021-05-01:
Hello what is the solution i am Interested .

DRSDavidSoft, 2021-05-03:
@ladyada This UF2 bootloader seems to be promising, I have to compare it to the Pi Pico and the products that you have!  The only downside is that it requires an actual Arduino with the 16u2 chip -- which sadly isn't present on many of the clones.

Personally, I use the CH340G USB-to-Serial chip in combination with the Optiboot bootloader, so the functionality to just use avrdude's read operation is super useful for the needs in my board.  Hopefully the IDE would also implement it.

<sup>P.S. It's so mindblowing seeing the CEO of Adafruit here, never thought it would happen 😄  Thank you for all the work you do @Adafruit, I truly appreciate the amount of good information that is on the Adafruit website.</sup>