arduino/Arduino #6146: Two protected areas in Stream.h

SummerSun, 2017-04-03:
Is it a typo?
```
class Stream : public Print
{
```
  **protected:**
```
    unsigned long _timeout;      // number of milliseconds to wait......
    unsigned long _startMillis;  // used for timeout measurement
    int timedRead();    // private method to read stream with timeout
    int timedPeek();    // private method to peek stream with timeout
public:
....
...
```
**protected:**
```
        long parseInt(char skipChar); // as above but the given skipChar is ignored
        // as above but the given skipChar is ignored
        // this allows format characters (typically commas) in values to be ignored

        float parseFloat(char skipChar);  // as above but the given skipChar is ignored
```

cousteaulecommandant, 2017-05-29:
This is intentional.  Apparently it used to be `private`, but it was changed to `protected` for optimization reasons (see
https://github.com/arduino/Arduino/commit/99f2a2755349784835130147e46cb61659b85893).
Maybe the comments should have been changed accordingly though.

Tip: use the **Blame** button when browsing a file on Github (https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Stream.h) to see which was the last commit to affect each line of the file.

matthijskooijman, 2017-05-30:
So the only change needed is to fix the comments. @cmaglie, perhaps you could fix this? Or make me collaborator on the sam/samd repositories too so I can push a simple fix?

per1234, 2017-07-18:
Closed by:
https://github.com/arduino/Arduino/issues/6464
https://github.com/arduino/ArduinoCore-samd/issues/244
https://github.com/arduino/ArduinoCore-sam/issues/37

closed 2017-07-18 by per1234