arduino/Arduino #6148: Using uint8_t causes hardware to stuck on both 1.8.1 and 1.8.2 with UNO

omersiar, 2017-04-03:
Hello All,

I am contributor to a specific hardware library for Arduino IDE (https://github.com/miguelbalboa/rfid/) which is cross-platform for other micro-controllers. We use standard data-types in examples to manage class-platform hardware.

The issue is not relevant to Library but IDE itself. For example: 
`
#include <EEPROM.h>

void setup() {
  pinMode(13, OUTPUT);
  for (int i = 0 ; i < EEPROM.length() ; i++) {
    EEPROM.write(i, 0);
  }
  digitalWrite(13, HIGH);
}

void loop() {
  /** Empty loop. **/
}`

If you open EEPROM clear example it just compiles and runs as expected.

but changing `int` data type to `uint8_t` it compiles but does not run. Workaround is using `int8_t`.
What do you think about this issue?

cmaglie, 2017-04-03:
You must always ensure that the datatype is large enough to hold the values you're handling.
In this case `uint8_t` is 8 bit wide, so it can hold values between 0 and 255 but `EEPROM.length()` is `1024` and your loop will run forever because `i` will be always less than `1024` (when `i` reaches `255` on the next increment it will rollover to `0`).

By using `int8_t`, instead, you go into another kind of error: the C/C++ language does not specify what 
 to do when a **signed** rollover happens, so the behaviour of the sketch in this case is **undefined** (this means that literally anything can happen, in this case it's just luck that the loop terminates, but I'm sure that the EEPROM has not been erased completely, so the overall behaviour of the loop is still wrong).

closed 2017-04-03 by cmaglie

omersiar, 2017-04-03:
I am total idiot, I am sorry, sir.