arduino/Arduino #6160: Attiny84 pinMapping issue

battosai30, 2017-04-07:
Hi,

I just observed an issue with Attiny84 (so I guess on 24/44 too) : 

http://drazzy.com/e/img/PinoutT84a.jpg

When I select "Damellis core" pin mapping, I get clockwise pin mapping in reality (I did not check if the inverse is the same).

per1234, 2017-04-08:
This is not the appropriate place to report this issue. ATTinyCore is a third party project and any issues should be reported to that repository:
https://github.com/SpenceKonde/ATTinyCore

It's not exactly clear to me what the issue you're reporting is. Are you saying that you think the label on the pinout diagram "Alternative pinout" for the "Counterclockwise (like ATTinyCore)" pin mapping is confusing as you would expect the "Clockwise (like damellis core)" pin mapping to be considered the "alternative"? It was [specifically decided](https://github.com/SpenceKonde/ATTinyCore/issues/50#issuecomment-202126223) that damellis mapping would be the "default", though that's also confusing since the ATTinyCore mapping is actually the default selection in the **Tools > Pin Mapping**.  It does seem like it would be more clear for the label on the diagram to match the text of the options in the **Tools > Pin Mapping** menu rather than using the vague term "Alternative pinout".

So if there is a legitimate bug or well explained suggestion for improvement then make it in a new issue at https://github.com/SpenceKonde/ATTinyCore and close this one. If you have any relevant replies to my comments here then it would be better to make them on the new issue report to keep all the conversation in the appropriate place and avoid spamming the followers of this repository. I'm following ATTinyCore so I will get an email from anything that happens over there.

closed 2017-04-08 by matthijskooijman

battosai30, 2017-04-08:
Oups ! Sorry I had two tabs open, on Arduino's git and AttinyCore's git and I wrote in the wrong one ...