arduino/Arduino #6163: arduino-1.8.2-linux64 doesn't install correctly, LMDE 2

jakeo25, 2017-04-08:
1.8.2 doesn't seem to install correctly.  I downloaded, moved to /opt, tar -xvf and then ran install.sh.  A bunch of errors, no desktop link, no program in start Menu.

Reverted to 1.8.1, same steps as above, works fine.

Using LMDE 2, Linux Mint 'Betsy' Cinnamon 64-bit
# uname -a
Linux 3.16.0-4-amd64 #1 SMP Debian 3.16.39-1 (2016-12-30) x86_64 GNU/Linux

Let me know if you need more detail

-Jake

per1234, 2017-04-08:
I'm going to take a wild guess that this is a duplicate of https://github.com/arduino/Arduino/issues/6116 and is fixed by https://github.com/arduino/Arduino/pull/6110.

matthijskooijman, 2017-04-08:
@jakeo25, could you see if this version of the install script works for you? https://github.com/artynet/Arduino/blob/92fd7232da8b3c511c29e587929b453f6fb697b1/build/linux/dist/install.sh

jakeo25, 2017-04-08:
MatthijsYes works great, 1.8.2 up and running.  I get  a couple warnings,  seems 'normal' for my distroAdding desktop shortcut, menu item and file associations for Arduino IDE...touch: cannot touch ‘/root/.local/share/applications/mimeapps.list’: No such file or directory/usr/bin/xdg-mime: 781: /usr/bin/xdg-mime: cannot create /root/.local/share/applications/mimeapps.list.new: Directory nonexistent done!

facchinm, 2017-04-10:
Closing as duplicate of #6116

closed 2017-04-10 by facchinm