arduino/Arduino #6165: Installing boards from the command line - howto ?

udoklein, 2017-04-08:
Today I added Travis testing to my library project (https://github.com/udoklein/dcf77/blob/travis/.travis.yml), https://travis-ci.org/udoklein/dcf77. I also want to include ARM /Arduino Due support. However the ARM boards are not predefined by the IDE. I found an article for including ESP8266 https://tzapu.com/automate-arduino-ide-esp8266-build-testing-travisci/. So the IDE does provide a command line interface to add board support from the command line like so

arduino --install-boards ...

Where can I find the proper parameters for this option to have Arduino Due available and how to I specifiy the Due as a compile target from the command line?

udoklein, 2017-04-08:
I also asked this in the Arduino Forum here: http://forum.arduino.cc/index.php?topic=468993

aido, 2017-04-08:
Hi Udo,

The following command will install the Arduino Due board:

`./arduino --install-boards arduino:sam`

and this command will build a sketch for the Arduino Due board:

`./arduino --verbose --verify --board arduino:sam:arduino_due_x <sketch name>`

udoklein, 2017-04-09:
Great. This raises the next question though. How / where do I learn these parameters for other boards?

aido, 2017-04-09:
Hi Udo,

The board names are in a `boards.txt` which on Linux can be found in the arduino install directory at:
`./hardware/arduino/avr/boards.txt`
or in the case of the manually installed SAM boards at:
`~/.arduino15/packages/arduino/hardware/sam/<version>/boards.txt`

On Windows this file is in:
`%LOCALAPPDATA%\Arduino15\packages\arduino\hardware`

closed 2017-04-09 by udoklein

luisangelcg007, 2020-06-05:
Here you can learn more about cmd command for arduino:
https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc

luisangelcg007, 2020-06-05:
> Great. This raises the next question though. How / where do I learn these parameters for other boards?

Here you can learn more about cmd command for arduino:
https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc