arduino/Arduino #6170: Running Arduino IDE under Windows (Installer or Windows Store) with ClearType causes the menus to display gibberish.

SarreqTeryx, 2017-04-08:
I'm not sure why, but shutting off ClearType fixes this problem, at the cost of text clarity

[image]

[image]

nitram509, 2018-10-11:
I would like to add some information.

I'm using Windows 10 Pro 64 bit (German version) with active Clear Type since years and never experienced such gibberish problem.
Maybe there is some issue overlapping, but clearly, it is not just ClearType setting alone.

Phoenix1o1, 2019-02-16:
I wish to add something to this too.
I am running Win 10 Pro Build 16299 x64, and as other reported turning Off cleartype fixes the issue with me too.
In my specific case the issue is (i think) due to the fact that I'm using a different default font systemwise (It's called B612). Reverting to windows default (i think it's called Segoe) fixes the issue.
Also, in the Setting windows of the IDE, the text can be read properly, so the issue is specific to menus only.
I suggest adding in the setting an option to choose the font to be used in the menus and allow the users to change it at will,  or investigating the issue more deeply.
I don't want to turn cleartype off any time I want to use the IDE and than back on when i'm finished.

Phoenix1o1, 2019-02-16:
[image]

Phoenix1o1, 2019-03-18:
@cmaglie do you have any update on this Issue? It could possibly be related to this #7839

facchinm, 2019-04-02:
It's surely related with #7839 ; a possible fix would be to override the menu fonts like this (https://github.com/arduino/Arduino/pull/8647/files#diff-0437e0831865dfac93630d3acc7321c9R43) but it would make them inconsistent with the rest of the system (better inconsistent that garbled anyway :stuck_out_tongue: )

cmaglie, 2019-04-08:
@Phoenix1o1 
It may be that the driver of your graphics card is not updated?

facchinm, 2019-04-18:
@SarreqTeryx @Phoenix1o1 @nitram509 I've investigated on the problem and found a solution that may fit.
By adding
```
-Dawt.useSystemAAFontSettings=on
```
to `arduino.l4j.ini` the fonts reappear as if ClearType was off, but just in the Arduino IDE and just for the menus.
Could you try if it also works for you? In this case we could add a line in the troubleshooting page (I would avoid applying this setting to everyone)

sandeepmistry, 2019-09-16:
I'm closing this for now due to lack of feedback, please re-open if needed with what @facchinm requested above.

closed 2019-09-16 by sandeepmistry

ammarchalifah, 2020-04-03:
I'm having this problem too. I'm using Windows 7, and I've tried the solution proposed by facchnim to add -Dawt.useSystemAAFontSettings=on to to arduino.l4j.ini. It doesn't work.

jeremyBass, 2020-08-10:
Just to add, I tried the 
`-Dawt.useSystemAAFontSettings=on`

and it did restore the font.  I am on the insider's build for win 10 build 20185.re_prerelease.200731-1415 although it should work for any win 10.