arduino/Arduino #6183: Board will not sync with com port

hconverse, 2017-04-13:
have admin access and still get the following message.  Cannot load to board, tried different com ports on computer and arduino software.  Using lilypad

Arduino: 1.6.7 (Windows 7), Board: "LilyPad Arduino, ATmega328"

Sketch uses 1,030 bytes (3%) of program storage space. Maximum is 30,720 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
avrdude: ser_open(): can't open device "\\.\COM5": The system cannot find the file specified.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

pedzed, 2017-04-30:
I have this issue as well. I can't work on the Arduino for a week now.

IDE Version: 1.8.2

Operating Systems:
- [elementaryOS](https://elementary.io) (desktop)
- elementaryOS (laptop)
- Windows 10

It used to work before, but suddenly it stopped working. I tried both with Arduino Uno and Mega 2560.

Console output in white:
```
Sketch uses 2730 bytes (8%) of program storage space. Maximum is 32256 bytes.
Global variables use 53 bytes (2%) of dynamic memory, leaving 1995 bytes for local variables. Maximum is 2048 bytes.
/opt/arduino-1.8.0/hardware/tools/avr/bin/avrdude -C/opt/arduino-1.8.0/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -carduino -P/dev/ttyACM1 -b115200 -D -Uflash:w:/tmp/arduino_build_154028/Program.ino.hex:i 
```

Followed with red/orangeish:
```
avrdude: Version 6.3, compiled on Jan 17 2017 at 11:00:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/opt/arduino-1.8.0/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/user/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM1
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
```

facchinm, 2017-05-02:
Hi @hconverse , did you select the correct port from Board/Port menu? Next time, before reporting, try updating the IDE to a more recent version because otherwise there is no way to understand if the bug has been solved in the meantime, thanks.

closed 2017-05-02 by facchinm