arduino/Arduino #6187: Serious code errors in Distance Detection with MaxSonar ultrasonic range finder example

BillWehner, 2017-04-15:
The mode() function in the example at http://playground.arduino.cc/Main/MaxSonar (see under "PW with mode filter") is very badly broken:
- the first while() always returns on the first pass
- there is an array bounds checking bug in the second while()
- the first if() uses & where && would be more correct
- for efficiency, "i" should always be incremented before the start of the outer while() regardless of whether count is 0 or not

Here's an improved version:

////////////////////////////////////////////////////////////////////////////////
/// bestSample() -- Return the "best" value in a sorted set of samples.
///
/// Returns the most common value if there is one, otherwise returns the median.
////////////////////////////////////////////////////////////////////////////////
uint16_t bestSample(uint16_t *x, uint8_t n)
{
    bool     bimodal       = false;     // set true if samples are bi or multimodal
    uint16_t selectedSample = 0;

    uint8_t maxCount  = 0;              // longest sequence found so far - 1

    // Search the samples for the longest sequence.

    for (uint8_t i=0; i < (n-1); i++)
    {
        // Find a sequence of identical values if there is one.

        uint8_t count = 0;              // number of identical samples - 1

        while (x[i]==x[i+1] && i<(n-1))
        {
            count++;
            i++;
        }

        // Keep track if this is the longest longest sequence so far.
        //
        if (count > maxCount)
        {
            selectedSample = x[i];
            maxCount = count;
            bimodal  = false;       // longest, so it's unique
        }
        //
        // If the sample set has multiple identical-length sequences, it's bi or multimodal.
        //
        else if (count!=0 && count == maxCount)
        {
            bimodal=true;
        }
    }

    // Return the median if there is no sequence or no unique sequence.

    if ( selectedSample==0 || bimodal )
    {
        selectedSample=x[(n/2)];
    }

    return selectedSample;
}

per1234, 2017-04-15:
The Arduino Playground is a publicly editable wiki. Maintaining the Playground content is the responsibility of the Arduino community, not the Arduino development team.

If you've tested your changes feel free to edit that page as you see fit. If you want to discuss the proposed changes with the community first then make a post on the Arduino Forum:
http://forum.arduino.cc/

facchinm, 2017-04-18:
Quoting @per1234 , the playground is totally user generated, so feel free to edit the page and fix the code if yours is tested and well-behaving :wink:

closed 2017-04-18 by facchinm