arduino/Arduino #6191: [Library Manager]: Please add BlueDot BME280 Library

BlueDot-Arduino, 2017-04-17:
https://github.com/BlueDot-Arduino/BlueDot_BME280

per1234, 2017-04-17:
Please read https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ and update your library to comply:

1. Add a [library.properties file](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#library-metadata).
2. Create a tag or release.

BlueDot-Arduino, 2017-04-18:
Thanks for the information. I created a library.properties file as well as a new release. I believe it is now in 1.5 format.

Rotzbua, 2017-04-18:
@BlueDot-Arduino There is no license information about your library, please add one :) .

BlueDot-Arduino, 2017-04-18:
Ops! Thanks again for the information. 
I made a few changes and I guess now the library should be complete! :)

agdl, 2017-04-20:
Done thank you!

closed 2017-04-20 by agdl