arduino/Arduino #6192: Can't change efuse and error write flash location 0x0

wenhuizhang, 2017-04-18:
Hi, I am using mega2560, and was trying to write board through `avrdude -p m2560 -c usbasp -P /dev/ttyUSB0 -B 115200 -e -u -Ulock:w:0x3f:m -U flash:w:./stk500boot_v2_mega2560.hex -Ulock:w:0x0F:m`, and when I was trying to change fuse through `avrdude -p m2560 -c usbasp -P /dev/ttyUSB0 -B 115200 -Uhfuse:w:0xda:m -Ulfuse:w:0xff:m  -Uefuse:w:0x05:m`, I have got :
```bash 

avrdude: set SCK frequency to 0 Hz
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801
avrdude: reading input file "0xda"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0xda:
avrdude: load data hfuse data from input file 0xda:
avrdude: input file 0xda contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of hfuse verified
avrdude: reading input file "0xff"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0xff:
avrdude: load data lfuse data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified
avrdude: reading input file "0x05"
avrdude: writing efuse (1 bytes):

Writing |                                                    | 0% 0.00s ***failed;  
Writing | ################################################## | 100% 0.06s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0x05:
avrdude: load data efuse data from input file 0x05:
avrdude: input file 0x05 contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xfd != 0x05
avrdude: verification error; content mismatch

avrdude: safemode: efuse changed! Was 5, and is now fd
Would you like this fuse to be changed back? [y/n] n
avrdude: safemode: Fuses OK (H:05, E:DA, L:FF)

avrdude done.  Thank you.

```
###The system I am using is :
```bash
avrdude: Device signature = 0x1e9801
avrdude> part 
>>> part 

AVR Part                      : ATmega2560
Chip Erase delay              : 9000 us
PAGEL                         : PD7
BS2                           : PA0
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

                         Block Poll               Page                       Polled
  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
```

facchinm, 2017-04-18:
Hi @wenhuizhang ,
which version of avrdude are you using? Version 6.3 we provide with the AVR core has been patched to avoid these kind of problems. In fact, some bits are marked as reserved and they always get read as 1 (or 0, it depends on the actual avrdude version), thus failing the verification step.

wenhuizhang, 2017-04-18:
Hi, @facchinm , thanks for your swift reply, I think I am using `avrdude: Version 6.0.1, compiled on Dec 16 2013 at 17:26:24`, I will try to update it to v6.3 and let you know how it goes.

per1234, 2017-07-06:
You need to use the AVRDUDE 6.3 that's included with the Arduino IDE. As facchinm said, It's been modified from the stock AVRDUDE. If you do this `brew upgrade` thing you'll get the stock version.

I sounds like you're not actually using the Arduino IDE. If so, it's possible to download only Arduino's build of AVRDUDE. The latest version at this time (6.3.0-arduino9) is available here:
- macOS: http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino9-i386-apple-darwin11.tar.bz2

I assume `brew upgrade` means you're using macOS but just in case:
- x86_64-linux: http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino9-x86_64-pc-linux-gnu.tar.bz2
- i686-linux: http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino9-i686-pc-linux-gnu.tar.bz2
- ARM Linux: http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino9-armhf-pc-linux-gnu.tar.bz2
- Windows: http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino9-i686-w64-mingw32.zip

per1234, 2017-08-04:
@wenhuizhang have you tried using the Arduino version of AVRDUDE? If so, did it solve the issue?

wenhuizhang, 2017-08-07:
Yep, it is solved, thanks so much :)

closed 2017-08-07 by facchinm

brenkem, 2019-05-22:
I have the same issue and want wo implement the latest avrdude version to buildroot.
For this I need a git repo with the latest release to build it on my own.

Is Arduino hosting its own repo for avrdude or what is the source where Arduino is getting its avrdude?

per1234, 2019-05-22:
@brenkem here's the source Arduino uses:
https://github.com/facchinm/avrdude
Here is the build script:
https://github.com/arduino/avrdude-build-script