arduino/Arduino #6193: Indentation of a blank line

septillion-git, 2017-04-18:
Dear all,

I searched the open issues and couldn't find an existing issue. Apologies in advance if there is already an issue about it.

In the IDE, when you are indented and you press return twice to insert a blank line, that blank line is not indented. This doesn't only look weird when you move your cursor, this is also pretty annoying if you made a blank line to insert a piece of code / to start coding there. Now you have to manually indent it again. I think this is one of the reasons the indentation of code from newbies is so mangled.

I tested it in some versions of the IDE (but I don't have all) and in 1.6.4 it still worked correct (aka, blank line is indented) but at least from 1.6.12 and up it mangles the indentation. Like I said, I find it annoying but more important, it really doesn't help for newbies to create nice indented code...

matthijskooijman, 2017-04-18:
From another perspective: It's common in the open-source code world to have empty lines without any identation, since the indentation is (in the resulting file) pointless (and usually labeled as "trailing whitespace" and often flagged as a problem). I  can see that, while editing, this whitespace could indeed be considered useful, especially if you make a bit of room for code and then later fill it in (instead of just writing it right away).

I'm not sure what the best approach is here, just wanted to point out another side to this issue.

matthijskooijman, 2017-04-18:
As for the reason why this is changed, I suspect that it might be related to the switch to RsyntaxTextArea, with the original TextArea preserving the indentation and RSTA cleaning it out.

septillion-git, 2017-04-18:
Yeah, from that perspective it does make sense but it shouldn't be harmful. And yeah, I sometimes make room for code to add (a couple of seconds 😛) later.

But in the case of Arduino, on the forum I see a lot of newbies who post code where the indentation is completely mangled. And I think this is a root cause of the mangled indentation. Newbies go back and forth through the code a lot and add bits and pieces along the way. And when they do, they are not used to indent themselves because the IDE does it for them if they write new code but it doesn't when they insert it.

So (in my opinion) two options to overcome this:
a) Indent blank lines (which in my opinion looks best)
b) Make the editor smarter and indent again when you add stuff to the line

carlosperate, 2017-04-18:
This kind of editor behaviour probably falls within the "personal preference" category, as different people will different opinions.

If one of the problems is having to press the tab key several times to reach the right position, what some editors do is to go to the previous code indentation level with a single tab keystroke, so that way there is no trailing whitespaces in empty lines, and it is still pretty painless to get to the right column.

septillion-git, 2017-04-18:
@carlosperate I don't completely agree in this case. It just mangles the indentation of newbies because they are not used to indent and just place the code where the IDE let's them. And I think last year the number of cases where a newbie had mangled code was bigger then before. I never really noticed the change in the IDE but I really think this is the root cause for newbies to have the code all over the place.

So for a newbie I think it would really help if the IDE just kept the indent. But because it's partly personal preference for the more experienced I would say that making this an option (in the preferences file, and default to keep indents) would be the nicest approach.

adabru, 2020-12-20:
Hi @septillion-git , does Arduino's autoformat function (Hotkey: Ctrl-T) solve your issue? It automatically indents each line correctly. I filed an issue to suggest adding an option to enable autoformating on saving a file: https://github.com/arduino/Arduino/issues/11090 . This way every time the user saves the file the indentation would be corrected.

septillion-git, 2020-12-21:
No, by far does it always fix it. Although it helps in basic situations.

But I would be against doing a auto format when saving. For two reasons:
- Users/newbies do not expect it and there code gets "mangled" when they save without knowing why.
- Auto formatter isn't good enough, there are situations, especially when code gets a bit more complicated, the auto formatter makes an even bigger mess.

Those two will probably be more irritating than useful for both for newbies and experienced users.

gelraen, 2020-12-24:
> Those two will probably be more irritating than useful for both for newbies and experienced users.

As an experienced user, I can already tell you that **not** having an option to autoformat-on-save is much more irritating, than the reasons you mention. With auto-format I don't have to:
* Argue about my formatting choices with teammates during codereview (becase now I don't have to make any such choices)
* See any whitespace diffs
* Even bother with wasting brain cycles on code formatting - simply hit Ctrl+S and it's done

Code formatters for C/C++ are good enough by now to not worry about "complicated" cases, I use `clang-format` at every possible opportunity, and didn't regret it even once.

However, I agree that auto-format on save might be very confusing if you're not expecting it, and probably should be off by default.