arduino/Arduino #6196: [IDE] Lose unsaved content when check "use external editor"

truebucha, 2017-04-19:
it is very distructive. I can't remember any IDE that could lose changes without promt.

cmaglie, 2017-04-19:
Please post instruction to reproduce the bug.

truebucha, 2017-04-21:
type some code into editor
press cmd+s
type more code and do nothing that make it autosave
go to settings and check the "use external editor" checkbox, press OK
What happened: editor reload file
What should happen: editor save changes before reload file

cmaglie, 2017-04-26:
Thanks for the detailed bug report!

Weird that nobody ever reported this, maybe is not so common that someone starts writing a sketch and in the middle of it swaps to external editor.

Yet it's still very dangerous, so I've pushed a fix on #6223 for the next IDE release.

PaulStoffregen, 2017-04-26:
I can confirm I've hit this several times, but it's never really bothered me.

truebucha, 2017-05-05:
My fault was I implemented new GPRS class at least 500 lines before change the editor =). that time it bothered me a lot.

PR #6223 (2017-05-16): Enabling "external editor" now requires to save all sketches.
Fix #6196

[image]
commit e500c7c4 (2017-05-16): Enabling "external editor" now requires to save all sketches.

Fix #6196
closed 2017-05-16 by cmaglie