arduino/Arduino #6198: Updating arduino AVR board

maddy2610, 2017-04-20:
I am using IDE 1.6.9. and after updating Arduino AVR Boards to v. 1.6.12, I can no longer upload to board. Going back to v. 1.6.11 and it works again.Tried this on Arduino Mega2560 and Nano. Same result.Any idea?  i have just bought Arduino products from this <a href="http://www.w11stop.com/buy-arduino-pakistan">Buy Arduino Pakistan</a>

facchinm, 2017-04-20:
Hi @maddy2610 ,
can you post the complete verbose output of compilation+upload?

per1234, 2017-04-21:
We tried to help you with this on the forum two weeks ago Maddy but you never bothered to respond. I'm guessing the problem is https://github.com/arduino/Arduino/issues/5173. Why do you ignore us when we try to help you and then make a duplicate post on Github? This is the third time you've done this. I'm starting to think you're just doing it to spam the link to your store.

References:
http://forum.arduino.cc/index.php?topic=468968
https://github.com/arduino/forum-issues/issues/225

http://forum.arduino.cc/index.php?topic=471126
https://github.com/arduino/forum-issues/issues/227

cmaglie, 2017-04-21:
> I'm starting to think you're just doing it to spam the link to your store.

Better to not buy from that store then since they are clones and the boards doesn't work :-)
BTW I edited the post to remove the link.

@maddy2610 please follow up to the forum threads and please don't advertise.

closed 2017-04-21 by cmaglie