arduino/Arduino #6201: problem with energia c2000 f2807

josele000, 2017-04-21:
i have a problem with the c2000. i press the button verify and my program its no compiling the message is the next:
**"C:\Users\JOSELE RENDON\AppData\Local\Temp\build1613906076524589072.tmp\sketch_apr19a.cpp", line 4: error #34: invalid line number
1 error detected in the compilation of "C:\Users\JOSELE RENDON\AppData\Local\Temp\build1613906076524589072.tmp\sketch_apr19a.cpp".

>> Compilation failure**
I already tried everything
helpme please my energia is the edition 0101E0017

facchinm, 2017-04-26:
Hi @josele000 , you should reopen the issue in the Energia issue tracker (https://github.com/energia/Energia), thanks

closed 2017-04-26 by facchinm

shakil1740, 2022-01-31:
> i have a problem with the c2000. i press the button verify and my program its no compiling the message is the next: **"C:\Users\JOSELE [...]

add 1 blank line at the beginning of your sketch.