arduino/Arduino #6222: [Library manager] Please add SdFat

awickert, 2017-04-26:
https://github.com/NorthernWidget/SdFat
This is a clone of the main component in the SdFat library belonging to @greiman. He has stated that he has moved on to new and exciting projects, but I really like his library, so have cloned it for inclusion in the IDE.
See https://github.com/greiman/SdFat/issues/27
Thanks!

per1234, 2017-04-26:
You need to create a release or tag that matches the `version` value set in library.properties. See https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ for more info. But before doing that please consider merging my pull request: https://github.com/NorthernWidget/SdFat/pull/1

awickert, 2017-04-26:
Just merged, but I see on the spec that "Data Storage" (without the quotes; that must have been my mistake) is a valid category. So, would it be better to change it to that than to leave it "Uncategorized"?:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

Also, the "release" is a date rather than a version number; this is what @greiman had. If this is not prefereable, I'm happy to declare this "v1.0.0" and change everything accordingly.

Once we decide on these, I'll create a tagged release. Thanks!

greiman, 2017-04-26:
I am moving on to a new library that supports FAT16/FAT32 and exFat.

The FAT16/FAt32 API will be compatible with the old SdFat.

I have finished the code and have done some testing.  I will be busy for several months so it will be a while before I release it.

awickert, 2017-04-26:
Thanks @greiman. I saw your new project, but it would be helpful to me to have a version of your current SdFat that is already built into the Arduino library manager. So could we do this, label it v1.0.0 (or whatever you'd like), and then update it when you finish?

greiman, 2017-04-26:
I will update SdFat with code from SdFat-beta and rearrange it for the library manager.

awickert, 2017-04-26:
Thanks @greiman. Then as I understand, you'll post the link to your library after the update, and at that point, my clone will become redundant and I'll delete it.

greiman, 2017-04-26:
I posted the new version of SdFat and an issue with it's location.

[https://github.com/greiman/SdFat](https://github.com/greiman/SdFat)

I did a release with the version 1.0.0 and set the version in library.properties to 1.0.0

q2dg, 2017-04-26:
Sorry for the offtopic...I only wanted to share my saddness because of the lack of any official response about this: https://github.com/arduino/Arduino/issues/3428

awickert, 2017-04-27:
@greiman Excellent, thanks very much! I'll close this, in that case, and remove my clone of your library (which I've really gotten some mileage out of -- so thanks for that as well!)

@q2dg I agree that it would be an improvement, but some more basic issues about how the built-in SD library prevent me from using. I'll send a +1 your way though.

closed 2017-04-27 by awickert