arduino/Arduino #6224: [Library manager] Please add SdFat

greiman, 2017-04-26:
https://github.com/greiman/SdFat

awickert cloned SdFat and posted a request for adding the clone to the library manager but it would be better to use my new version of SdFat.

See this issue for awickert's request.
https://github.com/arduino/Arduino/issues/6222

cmaglie, 2017-04-28:
Hi @greiman!

there is an executable inside the library, by policy we do not allow any executable to be inside a library:

```
JOB 947 - 2017/04/28 12:49:42 Scraping https://github.com/greiman/SdFat.git
JOB 947 - 2017/04/28 12:49:52 Checking out tag: 1.0.0
JOB 947 - 2017/04/28 12:49:52 Error syncing library: *.exe files found, skipping
```
it looks like that the detected .exe is here:
https://github.com/greiman/SdFat/tree/master/extras/AnalogBinLoggerExtras/bintocsv

may you remove it and re-tag the release?

greiman, 2017-04-28:
I removed the windows  bintocsv.exe and made release 1.0.1.

cmaglie, 2017-04-28:
Great, the library is now online!

closed 2017-04-28 by cmaglie