arduino/Arduino #6227: IDE displays #pragma message output as if it were an 'error', which it is not.

kriegsman, 2017-04-27:
The Arduino IDE visually renders the output of "#pragma message" as if it were an _error_.  When an Arduino library includes any #pragma message output (e.g. just to include version information to assist in debugging, as 'FastLED' does), end users are confused about what they see as "an error message", when no error condition actually exists.

[image]

This rendering style is appropriate for #pragma error output, but is not appropriate for #pragma message output.  #pragma message output should probably either be plain-text, the same as all other compiler output, or rendered in some other way that is visually different from an actual error, e.g., perhaps light blue instead of red.  I think my recommendation would be to have it render like ordinary compiler output, e.g., plain white.

facchinm, 2017-05-02:
Hi @kriegsman ,
in fact `#pragma ` statements are printed to `stderr` and this makes them orange. In fact, also `avrdude` output and warnings are orange since they both print to `stderr` too.
Solving the pragma specific problem is fairly easy with a filter but it would be nicer to be able to distinguish between an actual error and a simple print to stderr.

RufusVS, 2020-02-13:
is there any form of workaround until they get around to fixing this?  Or is it unfixable?   We are now in 2020 and the issue still exists.

highgear-me, 2024-08-31:
> Warning removal tworkshyd/V2Codebase#31

2024 and still there