arduino/Arduino #6229: Play melody page mistake?

grjonjon, 2017-04-27:
In "Play melody" page there are two code examples for producing tones on a passive piezo buzzer.
However, in the 1st example the user is instructed to use a "PWM pin" of the Arduino for producing the tone, which is pretty restricting. By going through the code I see that the tone is produced in software by turning that pin HIGH and LOW at predetermined time intervals; this can be done with a "non-PWM" pin, and to my opinion, more correctly so. After all, the used pin isn't used as a PWM out.
Correct me if I'm wrong.

Rotzbua, 2017-04-28:
Can you provide the link?

If I remember correctly, the pins have different timings to change the state and the PWM was the fastest pin..

facchinm, 2017-05-02:
The link should be https://www.arduino.cc/en/Tutorial/PlayMelody . 
In fact the first example doesn't need a PWM pin, while the second code (which uses `analogWrite` ) does. I believe that the pin was chosen that way to switch easily between the two examples; however the comments on the first example should be slightly changed.

tigoe, 2019-10-01:
Related to this, there appears to be an issue with `tone()` on the SAMD boards. When you play a tone of frequency <1, it freezes up. In the `playMelody()` example, for example, on a nano 33 Iot, it plays the first five notes of ["shave and a haircut"](https://www.youtube.com/watch?v=oSF9lMNjDTQ), then stops. If you change the sixth note, which is currently 0, to 1, then it plays the whole tune, "shave and a haircut, two bits". 

I haven't tried to replicate this on the other architectures recently, but I know that it worked on the Uno as of version 1.8.9.

per1234, 2020-09-29:
I seems that [PlayMelody tutorial](https://web.archive.org/web/20200720124329/https://www.arduino.cc/en/Tutorial/PlayMelody) was removed completely from arduino.cc sometime after 2020-07-20. Since nothing remains to fix, I'll go ahead and close this.

The bug reported by @tigoe was fixed by https://github.com/arduino/ArduinoCore-samd/pull/520

closed 2020-09-29 by per1234