arduino/Arduino #6234: [Library Manager] Please add GuaraTeca_OBR

JoaquimFlavio, 2017-04-29:
https://github.com/JoaquimFlavio/GuaraTeca_OBR

per1234, 2017-04-29:
This is the 4th issue you've opened about this library and you still haven't followed my instructions from the first one.

JoaquimFlavio, 2017-04-29:
What are the problems in the library?

per1234, 2017-04-29:
>What are the problems in the library?

As I already told you in https://github.com/arduino/Arduino/issues/6231 and https://github.com/arduino/Arduino/issues/6211, you need to create a release or tag that matches the `version` value in your library.properties file.
>The tag "version" follow the norms in the manual for Arduino!!!!

As you can see here:
https://github.com/JoaquimFlavio/GuaraTeca_OBR/tags
you have not created any releases or tags.

JoaquimFlavio, 2017-04-29:
Okay, I made the changes to the version tags!
I was confusing the version tags of GitHub with that of the file "library.properties"
Thanks for help!!

cmaglie, 2017-05-01:
Done!

closed 2017-05-01 by cmaglie