arduino/Arduino #6236: IDE V182 breaks SoftI2C-Master library

emrtech, 2017-04-29:
I recently upgraded from V1.65 to V1.82 and noticed, the projects no longer link properly with the SoftI2C-Master library. Up to V1.6.10 they work properly. In V1.82 not even the examples work anymore. When I get back to my PC I will document the type of error messages that were generated. It is a long list of identical errors. I just don't remeber right now. If you load one of the examples you can see for yourself. Anyways, I thought you may want to be aware of this issue. 

Thank you.
Regards,
Gerhard

facchinm, 2017-05-02:
SoftI2CMaster from https://github.com/todbot/SoftI2CMaster compiles correctly, while the old version fails with a lot of `SoftI2CMaster.h:502: undefined reference to ...` .
Please update the library from that github link and reopen the issue if the problem persists, thanks.

closed 2017-05-02 by facchinm

emrtech, 2017-05-02:
Desr Martino,

Thank you very much for your help. I was not aware of a new version. I  
did look but somehow, I didn't see it.

I will try it out today.

Best Regatds,
Gerhard