arduino/Arduino #6240: [Library Manager] Please add LM35

ggzucco, 2017-04-30:
https://github.com/ggzucco/LM35/tree/1.0

Thanks

per1234, 2017-04-30:
You need to add a library.properties file to your repository:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Details on this the library.properties file here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

ggzucco, 2017-04-30:
I uploaded the library.properties

cmaglie, 2017-05-01:
Tag 1.0 doesn't have the library.properties file, you probably created the file after making the release.

https://github.com/ggzucco/LM35/tree/1.0

a quick fix is to delete the tag 1.0.1 and make it again on master.

ggzucco, 2017-05-01:
ok, I do it =)

https://github.com/ggzucco/LM35

cmaglie, 2017-05-02:
Done!

closed 2017-05-02 by cmaglie