arduino/Arduino #6247: Serial.print() reference documentation is missing conditional function

thedub2001, 2017-05-01:
Hello I just discovered the the Serial.print() function can be used with conditional testing :
```
Serial.print(test ? "test=false" : "test=true");
```
If the "test" variable is true then the output will be : 
```
test=true
```

If the "test" variable is false then the output will be : 
```
test=false
```
I just thought this function could appear somewhere in the Serial.print() reference documentation

facchinm, 2017-05-02:
Well, it's not only about Serial.print , the ternary operator is allowed in any function if the "arguments" type is correct. However, it could be surely documented, but I'd prefer waiting until we merge some of [these](https://github.com/arduino/Arduino/pulls?q=is%3Apr+is%3Aopen+label%3A%22Print+and+Stream+class%22) PRs

NeatNit, 2017-05-14:
I think the `? :` operator is a common enough source of confusion to warrant a mention in the reference. I think it would fit nicely under "Further Syntax":

[image]

juanpablojofre, 2017-05-22:
I'm with @NeatNit!!
The ternary operator should be documented as part of the syntax of the language.
Where I disagree with @NeatNit is in the location.
Since it's an operator, it should be listed as such.
I would suggest a new header: Conditional Operator
and the entry underneath as:
- ? : (ternary operator)

NeatNit, 2017-05-22:
It would be lonely all by itself under that header. 😢

cousteaulecommandant, 2017-05-29:
Definitely not under "Further syntax"; this is just a plain old operator.  Maybe under "Boolean operators".

feikname, 2017-05-30:
Why not under control structures? The ternary operator is esssentially a one-line alternative to `if {} else { }` blocks.

cousteaulecommandant, 2017-05-30:
> Why not under control structures?

I'd rather not, since it's quite different from a control structure in how it's used – the statements in that section are typically used with blocks of code, which doesn't work with `? :`.
However it might be possible to put the documentation of `? :` as a section of the documentation for `if` as is done with comparison operators (and if not, I'd definitely link to `? :` on the "See also" section of `if`).

thedub2001, 2017-06-04:
I found another strange operator and I can't find documentation, perhaps it could be gathered at the same place in future documentation : 
```
for (auto dataobj : data){
    Serial.println(dataobj.key);
}
```
_example from https://bblanchon.github.io/ArduinoJson/doc/decoding/_

feikname, 2017-06-04:
@thedub2001 It's a [range-based `for` loop](https://stackoverflow.com/a/35490271).

cousteaulecommandant, 2017-06-17:
> `for (auto dataobj : data)`

I'd classify that as "advanced C++ *'black magic'* that is out of the scope of the Arduino documentation".
In fact all the code in that page uses quite a lot of C++ mechanisms such as references, templates, and custom classes, which are also seen in the internal Arduino code but are usually opaque to the user.

per1234, 2019-02-05:
Moved to https://github.com/arduino/reference-en/issues/519

closed 2019-02-05 by per1234