arduino/Arduino #6250: Arduino IDE infected

elyssam13, 2017-05-01:
Hi,
I just tried to download the lastest version of Arduino IDE and my antivirus is showing that it is infected. I am not allowed to have infected software on my work computer but need to use Arduino. Hopefully someone can resolve this issue.

facchinm, 2017-05-02:
Hi @elyssam13, where did you download the IDE from? The only place is https://www.arduino.cc/en/Main/Software . Double check the "checksums" to make sure the download is not corrupted. If the problem persists, please write here your computer specifications (Antivirus version, Windows version etc..)

elyssam13, 2017-05-02:
Hi @facchinm, yes that is where I went to download IDE from. Could you tell me how to check the "checksums". I am currently not allowed to have the download in any form on my computer. This is the message I am getting from symantec endpoint protection antivirus. I have windows 7 enterprise. I am not super computer savvy so I don't know what other computer specifications you would like. Just let me know and I'll tell you. 
[arduino error.docx](https://github.com/arduino/Arduino/files/971574/arduino.error.docx)

PaulStoffregen, 2017-05-02:
The many anti-virus programs occasionally have false positive detection for non-infected files.  Usually the vendor fixes the problem after a few days, on their next update.

I downloaded arduino-1.8.2-windows.exe from the website, just now.  The MD5 and SHA512 match.  I also did a full compare of the file against a copy I downloaded on March 22.  They are identical.

This file is almost certainly **not** infected with malware.  The problem is almost certainly a bad update to your anti-virus software.  This sort of bad update is a regular problem.  It's so common that anti-virus vendors have web pages for reporting false positives.

facchinm, 2017-05-03:
The "problem" found by Synantec Endpoint is [WS.Reputation.1](https://www.symantec.com/security_response/writeup.jsp?docid=2010-051308-1854-99) which is not an indication of a virus but a reputation system. I contacted Symantec to ease the whitelisting :slightly_smiling_face:

closed 2017-05-03 by facchinm