arduino/Arduino #6259: KeyboardMessage example code broken

gilesorr, 2017-05-05:
I just downloaded the latest Arduino IDE (overlaid with PJRC's Teensy code), version 1.8.2.  I wanted to test the Teensy++ as a USB Keyboard, so I went to File > Examples > 09.USB > Keyboard > KeyboardMessage.  The same code can be seen here: https://www.arduino.cc/en/Tutorial/KeyboardMessage

This code will not compile:

In file included from /home/giles/arduino-1.8.2/examples/09.USB/Keyboard/KeyboardMessage/KeyboardMessage.ino:23:0:
/home/giles/arduino-1.8.2/libraries/Keyboard/src/Keyboard.h:25:17: fatal error: HID.h: No such file or directory
 #include "HID.h"
                 ^
compilation terminated.
Error compiling for board Teensy++ 2.0.

The problem appears to be that "Keyboard.h" is now unneeded and unsupported?  I commented out the '#include' statement, and the code both compiled and worked.

Keyboard.h appears in all keyboard examples, and Mouse.h appears in all mouse examples and causes a very similar compile error.  This suggests that this is a widespread problem in the example code - it's a little disconcerting for beginners.

Can someone please fix the example code?  Thanks.

facchinm, 2017-05-05:
Hi @gilesorr ,
I believe that Teensy core uses a different USB implementation than standard AVR core, so the standard examples sketches are incompatible. You need to select the examples from "Teensy" menu and select the correct USB type from "Boards" menu. @PaulStoffregen correct me if I'm wrong :smile:

closed 2017-05-05 by facchinm

PaulStoffregen, 2017-05-05:
Yup, that's right.  There's also Teensy-specific examples in File > Examples > Teensy > USB_Keyboard.

When Arduino first started supporting USB keyboard (early 2013), I added Arduino's API to Teensy's USB keyboard object (first published in 2009), so sketches using Arduino's way would work.

But since then, Arduino's USB keyboard stuff has changed.  I will work on updating Teensy's API for compatibility.  Likely time frame for this update is July 2017.

gilesorr, 2017-05-05:
My mistake - my apologies!

Now that I've taken a quick look at the Teensy's examples ... the Arduino
ones appear to be more extensive and more straight-forward, so it would be
great if they were supported "out of the box" on the Teensy.  Thanks.

PaulStoffregen, 2017-06-02:
Just a quick followup: I've added empty header files, so these Arduino examples will compile and work on Teensy.  https://github.com/PaulStoffregen/cores/commit/a821a45c54219b85e2e38cefb2c3701a009c7e4d  https://github.com/PaulStoffregen/cores/commit/42972a31af8491ce23c1542b9e2409925bdd58de  These will be in the next Teensyduino software release, coming in mid-June.

The actual Teensy core library code is already fully compatible, when Keyboard or Mouse is selected in the menu.  The only issue was the need for these empty header files.

inventorman, 2019-08-22:
I have the same problem uploading my code which includes "Keyboard.h" on my digispark board.
Could you please help me?

oqibidipo, 2019-08-24:
DIgispark has its own keyboard library:

https://digistump.com/wiki/digispark/tutorials/digikeyboard

https://github.com/digistump/DigisparkArduinoIntegration/blob/master/libraries/DigisparkKeyboard/examples/Keyboard/Keyboard.ino
```
#include "DigiKeyboard.h"

void setup() {
  // don't need to set anything up to use DigiKeyboard
}

void loop() {
  // this is generally not necessary but with some older systems it seems to
  // prevent missing the first character after a delay:
  DigiKeyboard.sendKeyStroke(0);
  
  // Type out this string letter by letter on the computer (assumes US-style
  // keyboard)
  DigiKeyboard.println("Hello Digispark!");
  
  // It's better to use DigiKeyboard.delay() over the regular Arduino delay()
  // if doing keyboard stuff because it keeps talking to the computer to make
  // sure the computer knows the keyboard is alive and connected
  DigiKeyboard.delay(5000);
}
```

inventorman, 2019-08-25:
Actually, the point is that I need to use "Keyboard.h" library. 
Here is a sample code. I wonder if you could test it for me. I really appreciate it...
`#include "Keyboard.h"

void typeKey(int key)
{
  Keyboard.press(key);
  delay(50);
  Keyboard.release(key);
}

/* Init function */
void setup()
{
  // Begining the Keyboard stream
  Keyboard.begin();

  // Wait 500ms
  delay(500);

  delay(500);

  Keyboard.press(KEY_LEFT_GUI);
  Keyboard.press('r');
  Keyboard.releaseAll();

  delay(300);

  Keyboard.print("notepad.exe");

  delay(300);

  typeKey(KEY_RETURN);

  delay(300);

  Keyboard.print("Hello World!");

  // Ending stream
  Keyboard.end();
}

/* Unused endless loop */
void loop() {}`

per1234, 2019-08-25:
@inventorman this is not an appropriate place to get help with your project. The place for that is the Arduino Forum. I'm sure we'll be able to explain things over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).