arduino/Arduino #6262: Developer mailing list subscription not working

tni, 2017-05-05:
http://forum.arduino.cc/index.php?topic=471089.0

tni, 2017-05-05:
An email to: developers+subscribe@bcmi-labs.cc
did eventually work.

https://www.arduino.cc/en/Main/ContactUs
needs to be fixed.

per1234, 2017-05-06:
cmaglie provided some information about this on the mailing list:

>we recently changed mail-server and the broken links are an unwanted
side-effect. We are now in the middle of moving back the list under the
arduino.cc domain but this is going to take some days to complete.
>
>In the meantime the *temporary* link to subscribe is:
>
>https://groups.google.com/a/bcmi-labs.cc/forum/#!forum/developers
>
>Sorry for the troubles, and thanks for your patience...

So it sounds like, at least for now, the information about the mailing list found on arduino.cc and [this repository's Wiki](https://github.com/arduino/Arduino/wiki/Development-Policy#developers-mailing-list) should not be updated.

cmaglie, 2017-05-29:
The mailing list is back to arduino.cc.

closed 2017-05-29 by cmaglie