arduino/Arduino #6263: IDE data shift issue '<<' works only until 8 bits shift

Attila1978, 2017-05-05:
This is working in IDE:

a = (EEPROM.read(4) * 65536) + (EEPROM.read(5) *256) + EEPROM.read(6);

but this not:

a = (EEPROM.read(4) << 16) + (EEPROM.read(5) <<8) + EEPROM.read(6);

Rotzbua, 2017-05-05:
No bug, works as suggested.

```EEPROM.read``` return uint8_t, now you bitshift 16/8 to left, result is 0000 0000 = 0

This

    a = (EEPROM.read(4) << 16) + (EEPROM.read(5) <<8) + EEPROM.read(6);

is equal to

    a = 0 + 0 + EEPROM.read(6);

Attila1978, 2017-05-05:
I understand what you say. when 'a' is an unsigned long it need to be a 32 bit variable maybe I need try in this way:

a = (EEPROM.read(4)) << 16 + (EEPROM.read(5)) <<8 + EEPROM.read(6); 

exactly I would like to get the 3 bytes in a 32 bit variable.

Rotzbua, 2017-05-05:
Correct, `a` on the left side is 32bit, but on the right side the functions return `uint8_t` (that's good way to declare integer, because the size is independent from architecture). So on the right side it is always only 8bit, see eeprom class: https://github.com/arduino/Arduino/blob/2bfe164b9a5835e8cb6e194b928538a9093be333/hardware/arduino/avr/libraries/EEPROM/src/EEPROM.h#L120.

So you have to typecast the values before you bitshift them.

```c++
a = ((uint32_t)EEPROM.read(4)) << 16 + ((uint16_t)EEPROM.read(5)) <<8 + EEPROM.read(6);

matthijskooijman, 2017-05-08:
As @Rotzbua points out (Thanks!), this is how C/C++ is expected to work (confusing as it might be...), so I'm closing this ticket.

closed 2017-05-08 by matthijskooijman