arduino/Arduino #6267: 1-8-3/ 1-8-2 Amtel ICE Programer

anmalo, 2017-05-06:
Hi I not canuse the AmtelStudio 7, then i have Downloaded the Amtel USB 7 for the ICE.
Installing Arduino IDE 1.8.2/1.8.3 can't make a Connection to the Amtel ICE.

Where is the Problem?
I can't finde it.

facchinm, 2017-05-08:
Are you trying to program an AVR or a SAMD board? You should select (from Tools/Programmers) the correct programmer type ( `Atmel-ICE (AVR)` in the first case or `Atmel-ICE` in the second) and then use "Upload using programmer" menu.
Next time you file a bug report, please attach the **verbose output** of the action you are running, thanks

anmalo, 2017-05-08:
Hi
Thank you.
The Output is:
>>>>>
E:\Arduino\hardware\tools\avr/bin/avrdude -CE:\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega328p -catmelice_isp -Pusb -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xDA:m -Ulfuse:w:0xFF:m 
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "E:\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : atmelice_isp
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2141

avrdude done.  Thank you.

Fehler beim Brennen des Bootloaders.
<<<<<
The Board: Arduino Duemilanove
CPU: ATmega328
Programmer: Amtel ICE Basic, AVR Port, to 6 PIN JSP/SPI, Usb Driver Studio 7
Board Powered with 5 Volt.

facchinm, 2017-05-08:
Oh, wow.. Can you try and take a look at Windows device manager to make sure that the ICE has been correctly recognized by the system and exposes as `VID 0x03eb` and `PID  0x2141` ?

anmalo, 2017-05-08:
-> Amtel 
          Amtel-ICE Data Gateway
Das Gerät "USB\VID_03EB&PID_2141&MI_01\7&287c4aeb&1&0001" erfordert weitere Installationen.
I see the device need another install, but not known what.
USB Driver are installed now the newst Version that i have found. (7)

anmalo, 2017-05-08:
Reinstalled all Driver now:
Das Gerät USB\VID_03EB&PID_2141&MI_01\7&287c4aeb&1&0001 wurde gestartet.
But not works.

facchinm, 2017-05-08:
Ok, it could be a driver incompatibility. Would you test [Zadig](http://zadig.akeo.ie/) to force using `libusb` on the programmer? You can safely revert to Jungo (or any other driver in use) with the same tool.

anmalo, 2017-05-08:
I have installes all 4 Driver but not works.
On Zadig i see 3eb and 2141 when i select the amtel ICE

Is this a incompatiblity from Windows 10 64 bit Creator ?

matthijskooijman, 2017-05-08:
I've seen similar problems with the Atmel ICE on Windows on a system with Atmel Studio installed, which we fixed by running some tool from the libusb download to install a libusb filter for the Atmel ICE usb device (I think there were 2 or 3 subdevices that could be separately selected). Looking at this Zadig tool, I think it should do the same thing.

@anmalo, in Zadig, did you select "libusb" for this Atmel ICE device? Or did you just check to see if it is present?

anmalo, 2017-05-08:
Hi,
it is present on all Drivers with 3eb and 2142

anmalo, 2017-05-08:
Thank you für Help, it works with usblib_32. 

But i must be deinstall the amtel USB driver 7.
And Manual connect to usblib_32

Cheers.

Testato, 2017-05-11:
so if we install AtmelStudio, it is not possible use "Upload using programmer" by Atmel-ICE on Arduino IDE ? 
And cannot possible resolve this problem by Arduino side ? 
(I do not think that Atmel will change something on it side)

facchinm, 2017-06-06:
@Testato to overcome this issue we should "adopt" Atmel's upload tool on Windows (abandoning `avrdude` or anything else based on libusb = :no_entry_sign:). Zadig is a fast and lean way to switch the driver in use so there should be no problem even if the Atmel driver is still installed (I'll double check next time I boot Windows anyway :wink: )

closed 2017-06-06 by facchinm