arduino/Arduino #6277: Leonardo: constructor with delay knocks out serial port

Koepel, 2017-05-11:
In a forum in the Netherlands, someone encountered a possible bug: [http://arduinoforum.nl/viewtopic.php?f=13&t=2533](url).
I can replicate the problem in Windows 10 and Ubuntu linux 17.04 with Arduino IDE 1.8.1 and 1.8.2 and with the Leonardo bootloader that came with version 1.6.8 and 1.8.2.

Below is a test sketch. After uploading, the serial port disappears and uploading once more is not possible. The delay() is needed and the parameter for the constructor is needed.

There is a workaround to be able to upload a sketch. I turn on the verbose output for upload in the preferences and when the Arduino IDE tries to upload then I press the reset button on the Arduino board.

```
class myLibT
{
  public:
    myLibT( char a);
};

myLibT::myLibT( char a) 
{
  delay(10);     // This delay together with the parameter kills the usb-serial.
}

myLibT serialwrecker( 0);   // the parameter is needed

void setup() 
{
}

void loop() 
{
}
```

facchinm, 2017-05-19:
I think it is related with https://github.com/arduino/Arduino/pull/5592 . In fact, since the static initializers are executed before `init()`, the timer responsible for `delay()` doesn't fire yet and the board simply hangs.
I'm launching @ArduinoBot on that PR so you can test the produced build and check if the problem disappears.

Koepel, 2017-05-19:
Tested with "arduino-PR-5592-BUILD-691-linux64.tar.xz" and the problem is gone. Thank you!

sandeepmistry, 2019-09-16:
Based on the last comment this is fixed now, closing.

closed 2019-09-16 by sandeepmistry