arduino/Arduino #6280: [Library Manager] Please add CopyThreads

jensh, 2017-05-12:
https://github.com/jensh/CopyThreads
CopyThreads: A lightweight threading library.
I had the idea some days ago and are still quite impressed, how well and easy to use this multi-threading implementation works :-). Hope to get some feedback from others!

jensh, 2017-05-12:
Just realized, that there is already https://github.com/arduino-libraries/Scheduler/ . I am unsure, if it makes sense to port the "copy to heap" idea into Scheduler or keep it as a alternative implementation.

cmaglie, 2017-05-12:
I've added the library to the library manager.

There are many cooperative-threading libraries out now. The only thing that I feel to suggest you is to adapt your API to be compatible with the Scheduler library as the other authors did, so the final user can benefit of the possibility to switch between different implementations if needed.

closed 2017-05-12 by cmaglie

jensh, 2017-05-15:
Thanks for the listing. The API from newest Version 0.1.0 is compatible to the Scheduler library.