arduino/Arduino #6285: Mega pin 15 (Rx3) ...  is not working...

cemtaskent, 2017-05-13:
Although pin 14 is  claimed to be OK for Rx... It does not work .....

matthijskooijman, 2017-05-14:
Your report is really low on details, there's absolutely nothing we can do to help you based on what you wrote. Furthermore, this bug tracker is for reporting bugs and feature requests in the Arduino software, not for getting help with making your own code or project work. I suggest you ask on the Arduino forums, and make sure to include more details on what you're trying to do and what your observations are.

closed 2017-05-14 by matthijskooijman

cemtaskent, 2017-05-14:
Yes, I've also written my problem on forum..... 
The problem is about softserial which  GSM library using. 

Softserial is using change interupts for Rx. 
In the library there is a warning... 

"Not all pins on the Mega and Mega 2560 support change interrupts, so only the following can be used for RX: 10, 11, 12, 13, 14, 15, 50, 51, 52, 53, A8 (62), A9 (63), A10 (64), A11 (65), A12 (66), A13 (67), A14 (68), A15 (69)".... 

So pin 15 can bu used for Rx as claimed, but I can't use it... 
I'm just looking for the reason....

Thanks....

oqibidipo, 2017-05-14:
Pins 14 and 15 (and 0) support pin change interrupts, but they have been deliberately left out  from the PCINT macros. The macros were originally part of the SoftwareSerial library where this kind omission made sense.

IMHO now that the macros are universally available for different purposes, they should be updated to include all PCINT pins.

[`hardware/avr/variants/mega/pins_arduino.h`](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/mega/pins_arduino.h#L85)
```
// A majority of the pins are NOT PCINTs, SO BE WARNED (i.e. you cannot use them as receive pins)
// Only pins available for RECEIVE (TRANSMIT can be on any pin):
// (I've deliberately left out pin mapping to the Hardware USARTs - seems senseless to me)
// Pins: 10, 11, 12, 13,  50, 51, 52, 53,  62, 63, 64, 65, 66, 67, 68, 69

#define digitalPinToPCICR(p)     ...
#define digitalPinToPCICRbit(p)  ...
#define digitalPinToPCMSK(p)     ...
#define digitalPinToPCMSKbit(p)  ...

```

cemtaskent, 2017-05-14:
So is it possible to map pin 14 and 15 in the library.... 
If so how ?..
Sorry I'm a beginner :)