arduino/Arduino #6287: Reference missing mention of constants A0, A1, ...

NeatNit, 2017-05-14:
As I mentioned along with some other stuff [on the forum](http://forum.arduino.cc/index.php?topic=477052.0), the analog input pin constants are not mentioned in the [Arduino reference](https://www.arduino.cc/en/Reference/HomePage) - not under [Constants](https://www.arduino.cc/en/Reference/Constants) and not in [analogRead](https://www.arduino.cc/en/Reference/AnalogRead).

But actually, I'm unclear on how they are used. I have an Uno and the constant A0 has the value 14. The example AnalogReadSerial uses this constant to read the value: `int sensorValue = analogRead(A0);`
However, the analogRead reference page says:

>pin: the number of the analog input pin to read from (***0 to 5*** on most boards, ***0 to 7*** on the Mini and Nano, ***0 to 15*** on the Mega)

So which is it? Is the reference page plain wrong, or are A0 and 0 equivalent for analogRead?

matthijskooijman, 2017-05-14:
`analogRead` supports two argument values: The analog channel number (0-something), or the pin number (e.g. 14+ on a Uno) that can also be passed to `digitalWrite`. Does that clarify?

NeatNit, 2017-05-14:
Yes, thank you... though pert has already clarified this [on the forum](http://forum.arduino.cc/index.php?topic=477052.msg3260554#msg3260554).