arduino/Arduino #6291: ArduinoJson memory leak

bstsoftorg, 2017-05-16:
Hi
script
DynamicJsonBuffer ox_jsonBuffer(64);
JsonArray& ox = ox_jsonBuffer.createArray();
uint32_t ix = 0;
void timerIsr2()//Демонстрация утечки памяти
{
  Serial.println("heap:"+String(ESP.getFreeHeap())+"***ix="+(String)ix);
  Serial.println("size:"+(String)ox.size());
  while(ox.size())
  {
    ox.removeAt(0);
  }
  Serial.println("size:"+(String)ox.size());
  JsonObject& ox_item = ox.createNestedObject();
  Serial.println("size:"+(String)ox.size());
  Serial.println("heap:"+String(ESP.getFreeHeap()));
  ix++;
}

********************** log
Server started
heap:30904***ix=0
size:0
size:0
size:1
heap:30904
heap:28920***ix=51
size:1
size:0
size:1
heap:28920
heap:26856***ix=101
size:1
size:0
size:1
heap:26856
heap:26856***ix=151
size:1
size:0
size:1
heap:26856
heap:22744***ix=201
size:1
size:0
size:1
heap:22744
heap:22744***ix=301
size:1
size:0
size:1
heap:22744
heap:22744***ix=401
size:1
size:0
size:1
heap:22744
heap:14368***ix=1298
size:0
size:0
size:0
heap:14368

facchinm, 2017-05-16:
Hi @bstsoftorg ,
this should be reported to [ArduinoJson repository](https://github.com/bblanchon/ArduinoJson). However, I think it's the "correct" behaviour since you don't destroy the DynamicJsonBuffer (see https://bblanchon.github.io/ArduinoJson/doc/memory/)

closed 2017-05-16 by facchinm

bstsoftorg, 2017-05-16:
DynamicJsonBuffer not be removed, it is reusable and memory leaks should not be!!!
It is a global variable.

bstsoftorg, 2017-05-16:
Yes you are right the thing is DynamicJsonBuffer when you call size()
Serial.println("heap:"+String(ESP.getFreeHeap())+"***ix="+(String)ix+"***size:"+(String)ox_jsonBuffer.size());
I see the picture
heap:30168***ix=27***size:548
heap:14744***ix=473***size:9468
then the question arises how to clean DynamicJsonBuffer?