arduino/Arduino #6293: Serial.print bug ?

Testato, 2017-05-16:
this code work only for an half, 
it print only `12`

```
void setup() {
Serial.begin(9600);
}
void loop() {
  Serial.print(123);
  exit(0);    
}
```

facchinm, 2017-05-16:
The report is missing "some" information... If the bug report refers to a UNO, it's quite normal that not all bytes are being written, since the 16u2 changes its behaviour based on the status of the 328 (and probably changes baud rate, too, I have no time to check).
However, adding a small delay between the `print` and `exit` should solve the issue.
Reopen if needed

closed 2017-05-16 by facchinm

bengtmartensson, 2017-05-16:
Just two comments:

1. Normal programming practice says that you cannot rely everything to be output until you flush or close the buffer/stream. It appears that [Serial.end](https://www.arduino.cc/en/Serial/End) should be used to close the stream.

2. exit is meaningless in the Arduino context.

matthijskooijman, 2017-05-16:
Actually, I think `exit()` might be calling `abort()`, which IIRC disables interrupts and then starts looping infinitely. Because interrupts are disabled, serial output stops. If you insist on using `exit()`, make sure to call `Serial.flush()` to write all serial output first.

MauroMombelli, 2017-05-16:
correct answer is then:

1. Normal programming practice says that you cannot rely everything to be output until you flush or close the buffer/stream. It appears that Serial.end should be used to close the stream.

2. exit() [...] disables interrupts and then starts looping infinitely. Because interrupts are disabled, serial output stops.

This happen because the Serial library is asynchronous.