arduino/Arduino #6294: Feature Request: sketch properties

orgicus, 2017-05-16:
Processing saves a sketch.properties which store minimal data about the Processing mode the sketch uses.

Would it be possible to have a similar sketch.properties file that instead stores the Board and maybe Port information ? 

This would simplify switching between different sketches that uses different boards

facchinm, 2017-05-16:
This feature has been requested multiple times (see https://github.com/arduino/Arduino/issues/5121 for example), right now there is no plan to implement it (in the desktop IDE) using additional files but we are evaluating other methods. [Create](create.arduino.cc) already handles it in a transparent way, maybe try giving it a spin :wink:

closed 2017-05-16 by facchinm

mirage335, 2018-05-30:
This is now possible. See https://github.com/arduino/Arduino/issues/1481 .