arduino/Arduino #6295: Add useful features

ubaldoufo, 2017-05-16:
The editor lack of some feature, useful. In order to importance:

smart list of all routines of a project/file with speed selection (like in visual studio and many other editor)
same of above but for variables, constants, etc...
a way to resend last compiled program to arduino module without recompile it
a way to stop the compilation 

Thank you for your time.

Aurelio92, 2017-05-16:
This is not an issue and it shouldn't be here. If you want these features fork the project, add them and make a pull request

matthijskooijman, 2017-05-17:
Uh, there's nothing wrong with filing a feature request here, though it is usually better to stick to a single feature per issue. However, there is already ongoing work for the first feature (search for "autocomplete"), IIRC there is already another issue about stopping compilation, and resending the last program without recompiling is probably not going to happen (if you really need that, you can manually call avrdude to do so). In that light, I'll go ahead and close this issue, but thanks for your feedback anyway!

closed 2017-05-17 by matthijskooijman

ubaldoufo, 2017-05-18:
Thank you. Have a nice day.