arduino/Arduino #6297: IDE cannot set pins D20 (A6) and D21 (A7) with pinMode

rajivtctech, 2017-05-17:
I had an application which used all the pins of a Nano. I could not get a digital output on A6 and A7 which I had set with

#define onePin 20
#define anotherPin 21

void setup(){
pinMode(onePin, OUTPUT);
pinMode(anotherPin, OUTPUT);
}
etc...
The IDE reports no error but
digitalWrite(onePin, HIGH);
or
digitalWrite(anotherPin, HIGH);
does absolutely nothing.

It all works up to D19 (A5). Not with A6 or A7.

Rajiv

matthijskooijman, 2017-05-17:
These pins are a bit weird - they are analog only. There is no digital circuitry present in the 328p chip for them. If you check out the  datasheet, you'll see these pins are listed as "ADC6" and "ADC7" only, as opposed to other pins like "PC0 (ADC0/PCINT8)".

rajivtctech, 2017-05-17:
Thank you for the prompt response! I thought this was a bug in the hardware
configuration files...

Thank you.

Rajiv

closed 2017-05-18 by matthijskooijman